DoosanRobotics/doosan-robot #16: Moveit Connection Issue

adrianwslee, 2019-07-12:
I've successfully launched

roslaunch dsr_control dsr_moveit.launch mode:=real host:=192.168.127.100 model:m1013

and I don't get any error messages.

However, when I try to use Motion Planning, joints do not move.

When I take a look at the teach pendant, it shows control has been moved successfully.

Is there any other thing I have setup to connect robot arm with Moveit?

doosan-robotics, 2019-07-12:
Hello, This is Doosan Robotics.
Please try the following workarounds.

1. Check the network bandwidth.
roslaunch dsr_control dsr_moveit.launch mdoe: = real host: = 192.168.127.100 model: = m1013
If you use the following command, the ROS Master must have an IP of the form 192.168.127.x.
Since our controller router uses bandwidth 137, you need to set Static IP Address in order to use bandwidth 127.
If the connection is successful, the robot will go into Servo on state and you will hear a mechanical sound that turns on the motor.

2. Check the log messages.
For some operations, an error may occur due to the joint limit of each axis, and it may not move.
Try a small move that will not cause a limitation error.

Please try the upper method and if the problem is not resolved, please contact github's Issue category or ros.robotics@doosan.com.

Thank you.

adrianwslee, 2019-07-12:
[image]

I tried #2 of your suggestions
The attached picture shows the error I'm getting when I try Plan & Execute

adrianwslee, 2019-07-12:
I don't know the reason, but I think this only happens when the arm is in home position. 

First, I tried moving joints using hand guide, but, somehow joints were locked.

So I tried moving joints using teach pendant, then moved with hand guide option.

After that when I started the launch file, it moved.

Is there any safety lock when the robot is at home position that you have to initiate a motion using teach pendant? If not, I think there is a problem

doosan-robotics, 2019-07-12:
When the robot travels a short distance too fast, it automatically shifts to the appropriate speed (limitation speed). The message 1214 indicates that a speed limit has occurred. We modified the message type about the log level and then committed again.
Thank you for your feedback.

closed 2019-07-15 by adrianwslee