DoosanRobotics/doosan-robot #17: Moveit Control Issue

adrianwslee, 2019-07-15:
[image]

Hello,

I have couple more questions.

1. I'm keep getiing "Completed trajectory execution with status SUCCEEDED" when the arm didn't move at all. 

The arm display(URDF/SRDF model) from rviz gets updated to the goal position whilst the actual arm position remains the same. Any solution? (See the attached terminal output)

I've tried moving various distances, restarting a launch file, and rebooting both laptop and controller. However, I am keeping getting SUCCEEDED status when the arm hasn't moved.
 
2. I'm not getting any other error messages (besides 1214 message) when the arm doesn't repond after I pressed Plan & Execute. 

(When I start the launch file, I do hear a mechanical sound that the arm has entered the Server-On mode)

Thank you for your help.

doosan-robotics, 2019-07-15:
I want to know your situation exactly to solve the problem.
1. Is the robot not moving at all with Moveit? Or does it move in certain situations?
2. Are you running in an ROS kinetic environment?
3. When launching the launch file, is the control transfer related pop-up output to Teaching Pandaunt?
4. Does the sample file or service call work normally?
ex) example file
rosrun dsr_example_py single_robot_simple.py
ex) service
rosservice call /dsr01m1013/motion/move_joint "pos: [0.0, 0.0, 90.0, 0.0, 90.0, 0.0]
vel: 0.0
acc: 0.0
time: 4.0
radius: 0.0
mode: 0
blendType: 0
syncType: 0"

adrianwslee, 2019-07-25:
I made a stupid mistake (typo). Sorry!

closed 2019-07-25 by adrianwslee