DoosanRobotics/doosan-robot #22: Build error in Kinetic

adrianwslee, 2019-09-19:
Hi, I'm getting catkin_make error
'''
/usr/bin/ld: skipping incompatible /home/nvidia/arm_ws/src/doosan-robot/dsr_control/../common/lib/kinetic/x86_64/libDRFL.a when searching for -lDRFL
/usr/bin/ld: cannot find -lDRFL
collect2: error: ld returned 1 exit status
doosan-robot/dsr_control/CMakeFiles/dsr_control_node.dir/build.make:206: recipe for target '/home/nvidia/arm_ws/devel/lib/dsr_control/dsr_control_node' failed
make[2]: *** [/home/nvidia/arm_ws/devel/lib/dsr_control/dsr_control_node] Error 1
CMakeFiles/Makefile2:6042: recipe for target 'doosan-robot/dsr_control/CMakeFiles/dsr_control_node.dir/all' failed
make[1]: *** [doosan-robot/dsr_control/CMakeFiles/dsr_control_node.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
'''

adrianwslee, 2019-09-19:
Btw I'm using aarch64 architecture. 

I guess I'm getting this error because I'm using Nvidia TX2...

Are you willing to support aarch64 architecture too?

doosan-robotics, 2019-09-19:
The ARM architecture is not currently supported.
There is no architectural support plan for ARM yet.

closed 2019-09-19 by doosan-robotics