DoosanRobotics/doosan-robot #30: ROS2 Support

SeungRyeol, 2019-12-13:
M시리즈 라인업과 추후 공개될 A시리즈에 대해 추후 ROS2 지원 여부가 궁금하네요. 현재 개발 단계 인지 궁금하며 릴리즈된다면 언제쯤인지 알고 싶습니다. 항상 감사합니다.

doosan-robotics, 2019-12-16:
A시리즈는 ROS1에서 사용가능하도록 곧 배포될 예정입니다. ROS2 지원은 현재 개발 시작 단계 전이라 구체적인 릴리즈 일정을 답변드리기 어렵습니다.    

[Question]
I'm wondering if ROS2 support will be available for M series lineup and A series that will be released later. I'm wondering if it's in development, and when it's released.

[Answer]
A series will be released soon for use on ROS1. 
ROS2 support is now in the early stages of development, making it difficult to answer specific release schedules.

doosan-robotics, 2020-01-15:
A시리즈(a0509) 모델에 대한 기능 추가가 완료되었습니다. 

The function addition for the A series (a0509) model is completed.

doosan-robotics, 2020-11-03:
ROS2 지원을  시작하였습니다. 
Started support for ROS2.

https://github.com/doosan-robotics/doosan-robot2

omddness, 2020-11-03:
혹시 DDS를 사용하셨다면 어떤 DDS 제품을 사용하셨는지 여쭤봐도 될까요?

doosan-robotics, 2020-11-03:
ROS2는 기본 데이터 처리 방식이 DDS 통신입니다.
두산로보틱스 ROS2 패키지는 ROS2 설치 시 기본 탑재된 DDS를 사용하였습니다.

closed 2021-03-04 by SeungRyeol