DoosanRobotics/doosan-robot #31: ROS2 Support

doosan-robotics, 2020-01-15:
ROS2가 배포된지 2년이 넘었습니다. 
ROS2 로의 전환이 메가트랜드 인가요? 지원을 언제 해야만 하나요?
ROS2에 대한 여러분의 생각을 자유롭게 남겨주시면 감사하겠습니다.

It has been more than two years since ROS2 was released.
Is the transition to ROS2 megatrends? When should we apply?
I would be grateful if you could leave your thoughts about ROS2.

SeungRyeol, 2020-01-17:
Most experts insist on moving to ROS2. The last version of ROS1 will also be released this May. The last version supports LTS, but I think ROS2 will continue to be developed, so I will eventually have to go to ROS2. (AWS Robomaker already supports ROS2 Dashing.)

This is just my idea. :)

Thank you.

doosan-robotics, 2020-01-17:
Thank you for sharing a good idea.
For more information on using the Doosan Robot ROS package on AWS, see the aws branch.

SeungRyeol, 2020-01-17:
I've already tested it using AWS, but it's hard to handle. I think I need to study more.

closed 2020-02-28 by doosan-robotics