DoosanRobotics/doosan-robot #35: License

elizabethpeers, 2020-01-24:
Can you please add a license to the code. I know you have this here and that it is cloned to use during the ROS installation. But legally no one can use or edit the code without a license attached to it. 

Please add a license so it can be used legally.

doosan-robotics, 2020-01-28:
I will add the license for the serial node.
Can you tell me which licenses should be added further?

elizabethpeers, 2020-01-30:
I think you'll want either an MIT or GNU licensed. I think GNU GPLv3 would be best because it means the code and repo can be used in almost anyway but distributing closed source version.

Look at choosealicense.com

It will help you decide which is appropriate for your company.

Kind Regards,

Elizabeth Peers

E: elizabeth.peers@my.jcu.edu.au

Undergraduate Student of James Cook University

Bachelor of Engineering

-Major Electronic and Electrical

-Minor Physics

Undergraduate Thesis: Machine Learning in Astronomy - Classification of Quasars

elizabethpeers, 2020-02-18:
have you made any progress on adding a license?

doosan-robotics, 2020-02-19:
We specified open source license in the README.md file.
[image]

closed 2020-02-21 by doosan-robotics