DoosanRobotics/doosan-robot #37: Error when running  $ roslaunch dsr_launcher single_robot_gazebo.launch

scottnortman, 2020-01-28:
...
[ INFO] [1580182332.669205000, 0.068000000]: Loading gazebo_ros_control plugin
[ERROR] [1580182332.669441500, 0.068000000]: GazeboRosControlPlugin missing <legacyModeNS> while using DefaultRobotHWSim, defaults to true.
This setting assumes you have an old package with an old implementation of DefaultRobotHWSim, where the robotNamespace is disregarded and absolute paths are used instead.
If you do not want to fix this issue in an old package just set <legacyModeNS> to true.

[ INFO] [1580182332.671241600, 0.068000000]: Starting gazebo_ros_control plugin in namespace: /dsr01
[ INFO] [1580182332.673083300, 0.068000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [/dsr01/robot_description] on the ROS param server.
[ERROR] [1580182332.790878500, 0.068000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/joint1
[ERROR] [1580182332.794291100, 0.068000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/joint2
[ERROR] [1580182332.798343900, 0.068000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/joint3
[ERROR] [1580182332.802171600, 0.068000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/joint4
[ERROR] [1580182332.805066100, 0.068000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/joint5
[ERROR] [1580182332.807891400, 0.068000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/joint6
[ INFO] [1580182332.829139600, 0.068000000]: Loaded gazebo_ros_control.
[ WARN] [1580182332.829673200, 0.068000000]: The default_robot_hw_sim plugin is using the Joint::SetPosition method without preserving the link velocity.
[ WARN] [1580182332.830283200, 0.069000000]: As a result, gravity will not be simulated correctly for your model.
[ WARN] [1580182332.830635000, 0.069000000]: Please set gazebo_pid parameters, switch to the VelocityJointInterface or EffortJointInterface, or upgrade to Gazebo 9.
[ WARN] [1580182332.831069600, 0.069000000]: For details, see https://github.com/ros-simulation/gazebo_ros_pkgs/issues/612
...

doosan-robotics, 2020-01-28:
Is the Gazebo simulator not running at all?

scottnortman, 2020-01-28:
The simulator is running:

[image]

But I do not know how to fix the errors

doosan-robotics, 2020-01-28:
This error is output because the gazebo pid value is not specified. The pid value is optional, so there is no problem with the simulation.
You can check it in the /dsr_gazebo/config/gazebo_ros_control_params.yaml file.

Please refer to the [link](https://answers.ros.org/question/293830/what-is-the-fix-for-no-p-gain-specified-for-pid-namespace-gazebo_ros_controlpid_gainsback_right_wheel_joint-ros-melodic/)

closed 2020-02-05 by doosan-robotics