DoosanRobotics/doosan-robot #38: Using Hand-E Gripper

SeungRyeol, 2020-01-31:
Robotiq 사의 Hand-E를 이용해 dsr_example의 예제를 테스트해보던 중 문제가 발생했습니다.
일단 워크셀 매니저를 통해 Hand-E 제품을 등록하고 티칭펜던트로 동작확인은 모두 마쳤습니다.
roslanch를 이용해 노드들을 실행시키면 아래와 같이 주기적으로 gripper에 대한 메시지가 출력됩니다.
```
[ INFO] [1580445714.434525742]: On Monitor State
[callback OnMonitoringStateCB] current state: (1) STANDBY
[ INFO] [1580445714.435173511]: On Monitor State
[callback OnMonitoringStateCB] current state: (3) SAFE_OFF
[ INFO] [1580445714.819421485]: On Monitor State
[callback OnMonitoringStateCB] current state: (1) STANDBY
[callback OnMonitoringModbusCB] 2F_AR_5: 2304
[callback OnMonitoringModbusCB] 2F_GS_5: 63744
[callback OnMonitoringModbusCB] 2F_PR_5: 0
[callback OnMonitoringModbusCB] 2F_FS_PRE_5: 0
[callback OnMonitoringModbusCB] 2F_SP_FO_5: 65534
[callback OnMonitoringModbusCB] 2F_PO_CU_5: 768
[callback OnMonitoringModbusCB] 2F_AR_5: 2304
[callback OnMonitoringModbusCB] 2F_GS_5: 63744
[callback OnMonitoringModbusCB] 2F_PR_5: 0
[callback OnMonitoringModbusCB] 2F_FS_PRE_5: 0
```

dsr_example 안의 gripper 폴더에 있는 3가지 파일들을 실행 시켰을 때 다음과 같은 에러가 발생합니다.

1. object_pick_and_place_simple.py
```
shin@shin:~$ rosrun dsr_example_py object_pick_and_place_simple.py 
_robot_id =dsr01
_robot_model =m0609
_srv_name_prefix =/dsr01m0609
_topic_name_prefix =/dsr01m0609
Traceback (most recent call last):
  File "/home/shin/catkin_ws/src/doosan-robot/dsr_example/py/scripts/gripper/object_pick_and_place_simple.py", line 79, in <module>
    robotiq_2f_move(0.4)
  File "/home/shin/catkin_ws/src/doosan-robot/dsr_example/py/scripts/gripper/object_pick_and_place_simple.py", line 26, in robotiq_2f_move
    srv_robotiq_2f_move(width)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 515, in call
    responses = transport.receive_once()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 727, in receive_once
    p.read_messages(b, msg_queue, sock) 
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 353, in read_messages
    self._read_ok_byte(b, sock)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 336, in _read_ok_byte
    raise ServiceException("service [%s] responded with an error: %s"%(self.resolved_name, str))
rospy.service.ServiceException: service [/dsr01m0609/gripper/robotiq_2f_move] responded with an error: 
shutdown time!
shutdown time!
shutdown time!
```

2. pick_and_place_simple.py
```
shin@shin:~$ rosrun dsr_example_py pick_and_place_simple.py 
_robot_id =dsr01
_robot_model =m0609
_srv_name_prefix =/dsr01m0609
_topic_name_prefix =/dsr01m0609
Traceback (most recent call last):
  File "/home/shin/catkin_ws/src/doosan-robot/dsr_example/py/scripts/gripper/pick_and_place_simple.py", line 84, in <module>
    robotiq_2f_close()
  File "/home/shin/catkin_ws/src/doosan-robot/dsr_example/py/scripts/gripper/pick_and_place_simple.py", line 27, in robotiq_2f_close
    srv_robotiq_2f_close()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 515, in call
    responses = transport.receive_once()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 727, in receive_once
    p.read_messages(b, msg_queue, sock) 
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 353, in read_messages
    self._read_ok_byte(b, sock)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 336, in _read_ok_byte
    raise ServiceException("service [%s] responded with an error: %s"%(self.resolved_name, str))
rospy.service.ServiceException: service [/dsr01m0609/gripper/robotiq_2f_close] responded with an error: 
shutdown time!
shutdown time!
shutdown time!
```

3. real_pick_and_place_simple.py
```
shin@shin:~$ rosrun dsr_example_py real_pick_and_place_simple.py 
_robot_id =dsr01
_robot_model =m0609
_srv_name_prefix =/dsr01m0609
_topic_name_prefix =/dsr01m0609
Traceback (most recent call last):
  File "/home/shin/catkin_ws/src/doosan-robot/dsr_example/py/scripts/gripper/real_pick_and_place_simple.py", line 85, in <module>
    gripper_send_data(init_data)
  File "/home/shin/catkin_ws/src/doosan-robot/dsr_example/py/scripts/gripper/real_pick_and_place_simple.py", line 24, in gripper_send_data
    srv_gripper_send_data(send_data)
NameError: global name 'srv_gripper_send_data' is not defined
```

답변 부탁드립니다.

doosan-robotics, 2020-01-31:
Robotiq Gripper의 경우 기본적으로 RTU를 지원하되, 자체 지원하는 Controller를 쓰면 TCP/IP 통신도 가능한 것으로 알고 있습니다. 어떤 통신방식을 쓰고계신지 알려주시면, 해당 문제를 확인해 보겠습니다.

SeungRyeol, 2020-01-31:
티칭펜던트의 사전설정모드버스를 통해 설정한 IP주소, 포트 번호를 통해 통신하고 있습니다

doosan-robotics, 2020-01-31:
Robotiq - 2f 예제는 시리얼 통신 기반으로 작성되어있습니다. Modbus TCP/IP를 위한 예제를 만들어 구현되는 대로 등록하도록 하겠습니다.

doosan-robotics, 2020-01-31:
ip = <Gripper IP> #String 값
port = 502

add_modbus_signal(ip, port, "ACT", DR_MODBUS_REG_OUTPUT, 0, 0)   #modbus signal **ACT** 등록 : action request & gripper option
add_modbus_signal(ip, port, "WID", DR_MODBUS_REG_OUTPUT, 1, 0) # modbus signal **WID**등록 : Gripper Position
add_modbus_signal(ip, port, "FOR", DR_MODBUS_REG_OUTPUT, 2, 0) # modbus signal **FOR** 등록 : Gripper Force & Gripper Speed

set_modbus_output("ACT", 2304) # action request = 0x09 & gripper options = 0x00 / Gripper 활성화 신호와 Gripper 이동 요청 명령을 1로 set
set_modbus_output("WID", 0) # Gripper Position 을 최소 위치로 변경 (Open) / Gripper options2 = 0x00 & gripper position = 0x00
set_modbus_output("FOR", 65535) #  Speed = 0xFF / Force = 0xFF

wait(1)

set_modbus_output("WID", 255) # Gripper Position을 최대 위치로 변경 (Close) / Gripper options2 = 0x00 & gripper position = 0xFF

위 코드를 이용하여 테스트 해보세요.

SeungRyeol, 2020-02-04:
다음과 같은 오류가 발생합니다. (ip설정은 하였습니다.)
```
shin@shin:~$ rosrun dsr_example_py pick_and_place_simple.py 
_robot_id =dsr01
_robot_model =m0609
_srv_name_prefix =/dsr01m0609
_topic_name_prefix =/dsr01m0609
Traceback (most recent call last):
  File "/home/shin/catkin_ws/src/doosan-robot/dsr_example/py/scripts/gripper/pick_and_place_simple.py", line 80, in <module>
    add_modbus_signal(ip, port, "ACT", DR_MODBUS_REG_OUTPUT, 0, 0) #modbus signal ACT 등록 : action request & gripper option
  File "/home/shin/catkin_ws/src/doosan-robot/common/imp/DSR_ROBOT.py", line 1735, in add_modbus_signal
    srv = _ros_add_modbus_signal(name, ip, port, reg_type, index, _value, slaveid)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 515, in call
    responses = transport.receive_once()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 727, in receive_once
    p.read_messages(b, msg_queue, sock) 
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 353, in read_messages
    self._read_ok_byte(b, sock)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 336, in _read_ok_byte
    raise ServiceException("service [%s] responded with an error: %s"%(self.resolved_name, str))
rospy.service.ServiceException: service [/dsr01m0609/modbus/config_create_modbus] responded with an error: 
```

doosan-robotics, 2020-02-04:
dsr_example/py/scripts/demo 디렉토리에 test_modbus.py 파일을 추가하였습니다. 
robot_model, ip를 각각 실행하고자 하는 로봇 모델과 modbus 장치의 ip로 변경 후 실행시켜주세요.
해당 코드 모드버스 통신 작동 확인하였습니다.

SeungRyeol, 2020-02-04:
같은 메시지가 뜨길래 출력되는 아래의 정보를 이용해

```
[callback OnMonitoringModbusCB] 2F_AR_5: 2304
[callback OnMonitoringModbusCB] 2F_GS_5: 63744
[callback OnMonitoringModbusCB] 2F_PR_5: 0
[callback OnMonitoringModbusCB] 2F_FS_PRE_5: 0
[callback OnMonitoringModbusCB] 2F_SP_FO_5: 65534
[callback OnMonitoringModbusCB] 2F_PO_CU_5: 768
[callback OnMonitoringModbusCB] ACT: 0
```
add_modbus_signal 과정없이 
```
set_modbus_output("2F_PR_5", 255)
set_modbus_output("2F_PR_5", 0)
```
이렇게 코드를 작성하였더니 정상 작동하였습니다.

도움주셔서 정말 감사합니다.

closed 2020-02-04 by SeungRyeol