DoosanRobotics/doosan-robot #39: serial_node_example build issue

bgyoo91, 2020-02-06:
The executable file for the serial communication example (named serial_node_example) cannot be found.
and I get some error in dsr_example_py / real_pick_and_place.py
NameError: global name 'srv_gripper_send_data' is not defined

And I want to know when melodic-version is released.

doosan-robotics, 2020-02-07:
Fixed issue with serial_node_example not building.
If you run the setup_serial_node script in the / common / servicepack directory and run catkin_make, the serial_node_example package will build normally.

ex)
cd <your_work_space> / doosan-robot / common / servicepack
./setup_serial_node
cd <your_work_space>
catkin_make

We also fixed an issue that prevented the service from running in the pick_and_place example.

The gripper example of the doosan-robot package is Modbus RTU based on serial communication. If you want to use Modbus TCP / IP based gripper, use add_modbus_signal.

There is no support for Melodic devel, and the Melodic version of the wiki will be deleted.

doosan-robotics, 2020-02-07:
serial_node_example가 빌드되지 않던 문제를 해결하였습니다.
/common/servicepack 디렉토리에서 setup_serial_node 스크립트를 실행하고, catkin_make를 수행하면 serial_node_example 패키지가 정상적으로 빌드될 것입니다.

ex)
cd <your_work_space> / doosan-robot / common / servicepack
./setup_serial_node
cd <your_work_space>
catkin_make

또한 pick_and_place 예제에서 서비스가 실행되지 않던 문제를 수정하였습니다.

doosan-robot 패키지의 그리퍼 예제는 serial 통신 기반의 Modbus RTU를 상정하였습니다. Modbus TCP/IP 기반 그리퍼를 사용하려면 add_modbus_signal 함수 등의 Modbus TCP/IP를 위한 함수를 사용하십시오.

Melodic버전은 지원 예정이 없으며, ROS wiki의 Melodic version 관련 내용은 삭제 예정입니다.

closed 2020-02-10 by doosan-robotics