DoosanRobotics/doosan-robot #40: trans 함수 관련 질문

bgyoo91, 2020-02-13:
이전 두산의 윈도우기반 프로그램에서 ROS 로 변경하고 있는데,
trans 함수는 사라진건가요?
해당 함수를 사용하고 싶은데 가이드 부탁드립니다.

doosan-robotics, 2020-02-14:
DRL(Doosan Robot Language) 로 작성한 프로그램을 ROS로 옮기는 작업을 하고 계시군요.
ROS에서 trans 함수가 사라진 것이 아니라 ROS에서 현재 지원하지 않는 것입니다.
조만간 ROS 업데이트에 사용할 수 있도록 반영하겠습니다.

doosan-robotics, 2020-02-14:
포팅 작업하시면서 trans 함수뿐만 아니라 추가적으로 필요한 함수가 있으면 본 게시판에 남겨 주시기바랍니다.

bgyoo91, 2020-02-17:
@doosan-robotics 
감사합니다.
관련하여 궁금한점이 있는데, roslauch 를 통해 로봇을 구동하면서 집적교시를 할 수 있는 방법이 있나요?
RobotMode service 에 Manual 과 Autonomous 가 이를 뜻하는건가요?

doosan-robotics, 2020-02-18:
Manual Mode : 로봇 LED가 푸른 색으로 점등
Autonomous Mode : 로봇 LED가 하얀 색으로 점등
두 Robot Mode 모두 모션 명령어가 사용 가능하지만, Manual Mode에서는 감속되어 작동하기 때문에 모션 명령어 사용 시에는 Autonomous Mode로 변경하는 것을 권장합니다.
직접 교시는 Manual Mode에서만 사용 가능합니다. 아래 Service Call을 통하여 Robot Mode의 변경이 가능합니다..

rosservice call /dsr01m1013/system/set_robot_mode "robot_mode: 0"   // Manual Mode로 변경
rosservice call /dsr01m1013/system/set_robot_mode "robot_mode: 1"   // Autonomous Mode로 변경

doosan-robotics, 2020-02-18:
DSR_ROBOT.py에 래핑된 set_robot_mode 함수를 사용하여 Robot Mode의 변경이 가능합니다.
set_robot_mode(ROBOT_MODE_AUTONOMOUS)  # Autonomous Mode로 변경
set_robot_mode(ROBOT_MODE_MANUAL) # Manual Mode로 변경

set_robot_mode 사용 예시는 dsr_example/py/scripts/simple의 single_robot_simple.py 파일을 참고하세요.

doosan-robotics, 2020-02-19:
trans 함수가 추가 완료되었습니다. 
master branch 소스를 받아서 사용하시길 바랍니다.

closed 2020-02-21 by doosan-robotics