DoosanRobotics/doosan-robot #41: Using Moveit

SeungRyeol, 2020-02-17:
안녕하세요.

현재 2.5kg 정도되는 그리퍼를 만들어 다음과 같이 셋팅을 하였습니다.

[image]
[image]

문제는 Moveit을 사용할 때 목표 지점까지 명령을 줘도 0.3초 정도 움직이다가 아래와 같은 메시지와 함께 멈추고 맙니다. (티칭펜던트상의 조그 사용 시, 잘 작동하지만 Moveit을 사용하면 어떤 동작을 해도 오류 메시지와 함께 멈춥니다.)
```
[ INFO] [1581930124.355384396]: Planning attempt 1 of at most 1
[ INFO] [1581930124.359989971]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1581930124.361676386]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1581930124.402283069]: RRTConnect: Created 4 states (2 start + 2 goal)
[ INFO] [1581930124.402342420]: Solution found in 0.041378 seconds
[ INFO] [1581930124.418945718]: SimpleSetup: Path simplification took 0.016473 seconds and changed from 3 to 2 states
[ INFO] [1581930124.454843131]: callback: Trajectory received
[ INFO] [1581930124.454870311]:   msg->goal.trajectory.points.size() =10
[ INFO] [1581930124.454881212]:   msg->goal.trajectory.joint_names.size() =6
[ INFO] [1581930124.454897162]: [trajectory] [00 : 0.000] -19.745  19.989   0.007  13.730   0.006  -0.003
[ INFO] [1581930124.454910656]: [trajectory] [01 : 0.283] -17.551  17.768   0.006  12.204   0.006  -0.003
[ INFO] [1581930124.454920239]: [trajectory] [02 : 0.400] -15.357  15.547   0.005  10.678   0.006  -0.002
[ INFO] [1581930124.454937596]: [trajectory] [03 : 0.489] -13.164  13.325   0.004   9.152   0.006  -0.001
[ INFO] [1581930124.454946672]: [trajectory] [04 : 0.564] -10.970  11.104   0.003   7.626   0.005  -0.000
[ INFO] [1581930124.454954569]: [trajectory] [05 : 0.632]  -8.776   8.882   0.002   6.100   0.005   0.001
[ INFO] [1581930124.454962433]: [trajectory] [06 : 0.709]  -6.582   6.661   0.001   4.574   0.005   0.002
[ INFO] [1581930124.454971487]: [trajectory] [07 : 0.799]  -4.389   4.439   0.001   3.048   0.005   0.003
[ INFO] [1581930124.454978705]: [trajectory] [08 : 0.916]  -2.195   2.218  -0.000   1.522   0.005   0.003
[ INFO] [1581930124.454987325]: [trajectory] [09 : 1.194]  -0.001  -0.004  -0.001  -0.004   0.005   0.004
[ INFO] [1581930124.478524858]: On Monitor State
[callback OnMonitoringStateCB] current state: (2) MOVING
[ INFO] [1581930124.714329153]: [callback OnLogAlarm]
[ INFO] [1581930124.714467740]:  level : 1
[ INFO] [1581930124.714508645]:  group : 2
[ INFO] [1581930124.714572474]:  index : 1214
[ INFO] [1581930124.714610059]:  param : [INFO]finalTime(2.191) is adjusted above User Input(1.194)
[ INFO] [1581930124.714641131]:  param : 
[ INFO] [1581930124.714724955]:  param : 
[ INFO] [1581930125.033644182]: On Monitor State
[callback OnMonitoringStateCB] current state: (3) SAFE_OFF
[ WARN] [1581930125.060216333]: [callback OnLogAlarm]
[ WARN] [1581930125.060301489]:  level : 2
[ WARN] [1581930125.060338186]:  group : 5
[ WARN] [1581930125.060371090]:  index : 7056
[ WARN] [1581930125.060427996]:  param : 
[ WARN] [1581930125.060469619]:  param : 
[ WARN] [1581930125.060513666]:  param : 
[ INFO] [1581930125.062819542]: On Monitor State
[callback OnMonitoringStateCB] current state: (1) STANDBY
[ INFO] [1581930125.062907141]: On Monitor State
[callback OnMonitoringStateCB] current state: (3) SAFE_OFF
[ WARN] [1581930125.085651755]: [callback OnLogAlarm]
[ WARN] [1581930125.085733278]:  level : 2
[ WARN] [1581930125.085783231]:  group : 5
[ WARN] [1581930125.085816159]:  index : 7056
[ WARN] [1581930125.085879091]:  param : 
[ WARN] [1581930125.085907366]:  param : 
[ WARN] [1581930125.085931169]:  param : 
[ INFO] [1581930125.088255169]: On Monitor State
[callback OnMonitoringStateCB] current state: (1) STANDBY
[ INFO] [1581930125.088340069]: On Monitor State
[callback OnMonitoringStateCB] current state: (3) SAFE_OFF
[ WARN] [1581930125.100238107]: [callback OnLogAlarm]
[ WARN] [1581930125.100328828]:  level : 2
[ WARN] [1581930125.100365655]:  group : 2
[ WARN] [1581930125.100405060]:  index : 9015
[ WARN] [1581930125.100434071]:  param : 
[ WARN] [1581930125.100462285]:  param : err=External_Force, mode=Reduced, limit force=150.000(N)[33.721(lbf)], current force=314.659(N)[70.738(lbf)]
[ WARN] [1581930125.100494908]:  param : 
[ WARN] [1581930125.104939692]: [callback OnLogAlarm]
[ WARN] [1581930125.105017331]:  level : 2
[ WARN] [1581930125.105054327]:  group : 5
[ WARN] [1581930125.105095114]:  index : 7056
[ WARN] [1581930125.105124534]:  param : 
[ WARN] [1581930125.105150699]:  param : 
[ WARN] [1581930125.105179052]:  param : 
[ INFO] [1581930125.107274409]: On Monitor State
[callback OnMonitoringStateCB] current state: (1) STANDBY
[ INFO] [1581930125.107362584]: On Monitor State
[callback OnMonitoringStateCB] current state: (3) SAFE_OFF
[ WARN] [1581930125.119735647]: [callback OnLogAlarm]
[ WARN] [1581930125.119807116]:  level : 2
[ WARN] [1581930125.119839400]:  group : 2
[ WARN] [1581930125.119864077]:  index : 9015
[ WARN] [1581930125.119897729]:  param : 
[ WARN] [1581930125.119958187]:  param : err=External_Force, mode=Reduced, limit force=150.000(N)[33.721(lbf)], current force=762.194(N)[171.348(lbf)]
[ WARN] [1581930125.119991266]:  param : 
[ WARN] [1581930125.124354572]: [callback OnLogAlarm]
[ WARN] [1581930125.124427237]:  level : 2
[ WARN] [1581930125.124463225]:  group : 5
[ WARN] [1581930125.124508466]:  index : 7056
[ WARN] [1581930125.124550764]:  param : 
[ WARN] [1581930125.124576932]:  param : 
[ WARN] [1581930125.124615250]:  param : 
[ INFO] [1581930125.126904423]: On Monitor State
[callback OnMonitoringStateCB] current state: (1) STANDBY
[ INFO] [1581930125.126996221]: On Monitor State
[ INFO] [1581930125.454985393]: Completed trajectory execution with status SUCCEEDED ...
[callback OnMonitoringStateCB] current state: (3) SAFE_OFF
[ WARN] [1581930125.513818751]: [callback OnLogAlarm]
[ WARN] [1581930125.513902390]:  level : 2
[ WARN] [1581930125.514129764]:  group : 2
[ WARN] [1581930125.514172990]:  index : 9015
[ WARN] [1581930125.514198675]:  param : 
[ WARN] [1581930125.514407511]:  param : err=External_Force, mode=Reduced, limit force=150.000(N)[33.721(lbf)], current force=1198.887(N)[269.521(lbf)]
[ WARN] [1581930125.514459265]:  param : 
[ WARN] [1581930125.519264245]: [callback OnLogAlarm]
[ WARN] [1581930125.519339904]:  level : 2
[ WARN] [1581930125.519372944]:  group : 5
[ WARN] [1581930125.519397857]:  index : 7056
[ WARN] [1581930125.519449758]:  param : 
[ WARN] [1581930125.519475705]:  param : 
[ WARN] [1581930125.519498626]:  param : 
```

M0609 모델을 사용 중이며, 2달전 그립퍼를 사용하지 않을 때에도 테스트할 때 제한문제로 멈춤이 있었습니다.
```
[ WARN] [1581930125.100462285]:  param : err=External_Force, mode=Reduced, limit force=150.000(N)[33.721(lbf)], current force=314.659(N)[70.738(lbf)]
```
이 부분을 보면 제한 문제 같은데, 추가로 더 설정해줘야하는 부분이 있나요?

doosan-robotics, 2020-02-18:
일단 그립퍼를 떼고 툴무게 값을 0으로 초기화한 후,
무부하 상태에서 
1) 임의의 자세에서 직접교시 버튼을 누릅니다. 
2)  로봇 홈포지션 자세에서 직접교시 버튼을 누릅니다.

1) 2)에서 로봇이 임의의 방향으로 흐르는지 체크바랍니다.

SeungRyeol, 2020-02-18:
답변감사합니다. 맨 마지막에 임의의 방향이라고하는게 어떤걸 말씀하시는 건가요?

doosan-robotics, 2020-02-18:
직접교시 버튼을 눌렸을 때 움직이지 않고 자세를 유지하는 것이 정상입니다.

doosan-robotics, 2020-02-18:
상기 TEST를 진행하시기 전에 Autonomous Mode로 전환하신 후, 진행해 보시길 바랍니다.
Manual Mode : 로봇 LED가 푸른 색으로 점등
Autonomous Mode : 로봇 LED가 하얀 색으로 점등
두 Robot Mode 모두 모션 명령어가 사용 가능하지만, Manual Mode에서는 감속되어 작동하기 때문에 모션 명령어 사용 시에는 Autonomous Mode로 변경하는 것을 권장합니다.

rosservice call /dsr01m1013/system/set_robot_mode "robot_mode: 0" // Manual Mode로 변경
rosservice call /dsr01m1013/system/set_robot_mode "robot_mode: 1" // Autonomous Mode로 변경

SeungRyeol, 2020-02-19:
답변 감사합니다. Autonomous Mode로 하니 작동 되네요. 추가적으로 셋팅에 대한 질문이 있습니다.

1. 그립퍼를 뗀 상태에서는 직접교시 버튼을 눌렸을 때 움직이지 않고 자세가 고정되지만 그립퍼를 부착한 상태에서는 무게(약 2.5kg) 때문에 축처지는 현상이 발생합니다. 그래서 직접교시 할 때는 손으로 받쳐주는 과정이 필요했구요. 정상적인 반응이 맞나요? 아니면 추가적인 셋팅이 필요한가요?

2. 그립퍼를 붙인 상태로 moveit을 사용하거나 그냥 python 상에서 movej() 함수 등을 이용해 움직일 때, 사전에 워크셀 매니저를 이용해 그립퍼 무게를 측정후 셋팅 및 적용을 했었는데 이게 옳은 절차 맞나요?

doosan-robotics, 2020-02-20:
[1번 질문 답변] 
그립퍼를 다시고 다시 그립퍼 셋팅을 하셨나요? 
하셨다면 하기 절차대로 다시 셋팅바랍니다.
[image]
1) 로봇이 설치된 공간제약이 없다면 Full Motion 을 선택합니다. 
2) Weight 에서 체크박스를 선택해제 후, 툴의 무게값을 직접 입력합니다.
3) Auto Measure 를 실시합니다. 
만약 상기 방법을 실시해도 계속 흐른다면 구매처를 통해서 정식으로 기술지원을 받으시길 권장드립니다.

[2번 질문 답변]
  네 맞습니다.

SeungRyeol, 2020-02-20:
기존에 그립퍼 설정을 지우고 알려주신 절차대로 진행 후 직접교시를 수행했는데 무게 때문에 떨어집니다. 혹시 문의할 곳을 알려주실 수 있으신가요?

SeungRyeol, 2020-02-20:
[image]
아이콘을 눌러 추가적인 설정을 하였더니 직접교시 버튼을 눌러도 떨어지지 않습니다. 감사합니다.

closed 2020-02-20 by SeungRyeol

doosan-robotics, 2020-02-21:
사용하시는 SW 버전이 어떻게 되시나요?

SeungRyeol, 2020-02-21:
M2.4.1 패치버전 사용하고 있습니다. 
워크셀 1.1 사용문제 때문에 2.4.1에서 패치 버전으로 업데이트 받았습니다.

doosan-robotics, 2020-02-21:
네 알겠습니다. 수고하세요.