DoosanRobotics/doosan-robot #43: M1013 issue

bgyoo91, 2020-02-27:
m1013 모델을 사용중입니다.
roslaunch dsr_launcher single_robot_rviz.launch model:=m1013 mode:=real host:=<ip>
를 이용해서 작업중인데, 일정시간동안 켜놓으면 (약 5분)
'따다따닥'소리 (로봇이 activate 될때 소리)가 나면서 LED가 청색으로 바뀝니다.

그 뒤에, 로봇제어 코드를 실행하면 (movej) 로봇이 가다 멈추다 가다 멈추다를 반복하다가
빨간 LED로 바뀌고 안전모드로 빠집니다.

**이 증상은 일정시간(5분)에 sleep 모드로 들어가는 의도된 행동같은데, 그냥 manual mode 로 변경되는것뿐인가요?**
**해제 혹은 코드상에서 이를 감지하고 원상태로 복구할 수 있나요?**

아래는 dsr_launcher log 입니다. (bold 부분이 해당 증상후 로그)

SUMMARY
========

PARAMETERS
 * /dsr01/dsr_joint1_position/joint: joint1
 * /dsr01/dsr_joint1_position/pid/d: 200.0
 * /dsr01/dsr_joint1_position/pid/i: 100.0
 * /dsr01/dsr_joint1_position/pid/i_clamp: 500
 * /dsr01/dsr_joint1_position/pid/p: 500.0
 * /dsr01/dsr_joint1_position/type: position_controll...
 * /dsr01/dsr_joint2_position/joint: joint2
 * /dsr01/dsr_joint2_position/pid/d: 250.0
 * /dsr01/dsr_joint2_position/pid/i: 100.0
 * /dsr01/dsr_joint2_position/pid/i_clamp: 500
 * /dsr01/dsr_joint2_position/pid/p: 500.0
 * /dsr01/dsr_joint2_position/type: position_controll...
 * /dsr01/dsr_joint3_position/joint: joint3
 * /dsr01/dsr_joint3_position/pid/d: 40.0
 * /dsr01/dsr_joint3_position/pid/i: 20.0
 * /dsr01/dsr_joint3_position/pid/i_clamp: 500
 * /dsr01/dsr_joint3_position/pid/p: 300.0
 * /dsr01/dsr_joint3_position/type: position_controll...
 * /dsr01/dsr_joint4_position/joint: joint4
 * /dsr01/dsr_joint4_position/pid/d: 40.0
 * /dsr01/dsr_joint4_position/pid/i: 0.0
 * /dsr01/dsr_joint4_position/pid/i_clamp: 500
 * /dsr01/dsr_joint4_position/pid/p: 100.0
 * /dsr01/dsr_joint4_position/type: position_controll...
 * /dsr01/dsr_joint5_position/joint: joint5
 * /dsr01/dsr_joint5_position/pid/d: 25.0
 * /dsr01/dsr_joint5_position/pid/i: 20.0
 * /dsr01/dsr_joint5_position/pid/i_clamp: 500
 * /dsr01/dsr_joint5_position/pid/p: 200.0
 * /dsr01/dsr_joint5_position/type: position_controll...
 * /dsr01/dsr_joint6_position/joint: joint6
 * /dsr01/dsr_joint6_position/pid/d: 25.0
 * /dsr01/dsr_joint6_position/pid/i: 20.0
 * /dsr01/dsr_joint6_position/pid/i_clamp: 500
 * /dsr01/dsr_joint6_position/pid/p: 300.0
 * /dsr01/dsr_joint6_position/type: position_controll...
 * /dsr01/dsr_joint_position_controller/joints: ['joint1', 'joint...
 * /dsr01/dsr_joint_position_controller/publish_rate: 20
 * /dsr01/dsr_joint_position_controller/type: position_controll...
 * /dsr01/dsr_joint_publisher/publish_rate: 150
 * /dsr01/dsr_joint_publisher/type: joint_state_contr...
 * /dsr01/dsr_joint_trajectory_controller/joints: ['joint1', 'joint...
 * /dsr01/dsr_joint_trajectory_controller/publish_rate: 20
 * /dsr01/dsr_joint_trajectory_controller/type: position_controll...
 * /dsr01/dsr_velocity_controller/angular/z/has_acceleration_limits: True
 * /dsr01/dsr_velocity_controller/angular/z/has_velocity_limits: True
 * /dsr01/dsr_velocity_controller/angular/z/max_acceleration: 6.0
 * /dsr01/dsr_velocity_controller/angular/z/max_velocity: 2.0
 * /dsr01/dsr_velocity_controller/base_frame_id: base_link
 * /dsr01/dsr_velocity_controller/cmd_vel_timeout: 0.25
 * /dsr01/dsr_velocity_controller/enable_odom_tf: False
 * /dsr01/dsr_velocity_controller/estimate_velocity_from_position: False
 * /dsr01/dsr_velocity_controller/left_wheel: ['front_left_whee...
[... 158 lines cut ...]
[INFO] [1582776580.401118]: Loading controller: dsr_joint_position_controller
[INFO] [1582776580.416896]: Loading controller: dsr_joint_publisher
[INFO] [1582776580.426883]: Controller Spawner: Loaded controllers: dsr_joint_position_controller, dsr_joint_publisher
[INFO] [1582776580.436730]: Started controllers: dsr_joint_position_controller, dsr_joint_publisher
process[dsr01/world_tf-5]: started with pid [7073]
process[dsr01/spawn_create_model-6]: started with pid [7127]
process[dmodel-7]: started with pid [7140]
[WARN] [1582776609.273579]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[dsr01/controller_spawner-1] process has finished cleanly
log file: /root/.ros/log/cb69b3a0-5909-11ea-8461-00012e93400b/dsr01-controller_spawner-1*.log
**[ INFO] [1582776880.264821075]: [callback OnLogAlarm]
[ INFO] [1582776880.264907976]:  level : 1
[ INFO] [1582776880.264985765]:  group : 1
[ INFO] [1582776880.265027732]:  index : 1038
[ INFO] [1582776880.265054018]:  param : 
[ INFO] [1582776880.265082732]:  param : 
[ INFO] [1582776880.265109566]:  param : 
[ INFO] [1582776880.267497275]: On Monitor State
[callback OnMonitoringStateCB] current state: (3) SAFE_OFF
[ INFO] [1582776880.287912205]: On Monitor State
[callback OnMonitoringStateCB] current state: (1) STANDBY
[ INFO] [1582776880.369411787]: On Monitor State
[callback OnMonitoringStateCB] current state: (3) SAFE_OFF
[ INFO] [1582776880.756358728]: On Monitor State
[callback OnMonitoringStateCB] current state: (1) STANDBY**

bgyoo91, 2020-02-27:
그리고 onLogAlarm 의 의미도 알고싶습니다.
[ WARN] [1582789611.029492359]: [callback OnLogAlarm]
[ WARN] [1582789611.029527010]:  level : 2
[ WARN] [1582789611.029536786]:  group : 2
[ WARN] [1582789611.029544963]:  index : 1209
[ WARN] [1582789611.029555873]:  param : 
[ WARN] [1582789611.029563906]:  param : 
[ WARN] [1582789611.029571799]:  param : 

메뉴얼대로라면, level2. 단순 기능 및 동작 오류로 인한 로봇이 정지된 상태
group 2. 하위 제어기(알고리즘) 인데,
error code 인 index 의 의미가 안적혀있습니다.
1209의 의미는 무엇이며, error code 마다의 의미도 알려주시면 감사하겠습니다.

doosan-robotics, 2020-02-28:
안전 문제로 인하여 Stay 상태에서 일정 시간(기본값 5분)이 지나면 자동으로 Servo Off가 됩니다. TP에서 해당 설정을 변경할 수 있습니다.
Setting - Idle Servo Off - Idele Servo Off를 비활성화하거나, Servo Off Robot Status after 항목의 값을 변경한 후 Confirm을 선택하세요.
[image]

doosan-robotics, 2020-02-28:
그리고 매뉴얼 상에 에러코드는 어떤 매뉴얼을 보셨는지 모르겠지만...
우선 답변드리면 1209에러는 입력된 목표 속도 (TASK)가 사용자 LIMIT을 벗어날을 때 나는 에러입니다.

bgyoo91, 2020-02-28:
@doosan-robotics 답변 감사합니다!
메뉴얼은 readme 에 적혀있는 Manual(KOR)을 보고있습니다.
http://wiki.ros.org/doosan-robotics?action=AttachFile&do=get&target=Doosan_Robotics_ROS_Manual_ver0.971_20200218A%28Kor.%29.pdf#page41

여기에 error code에 대한 정보는 없는것 같습니다.

doosan-robotics, 2020-02-28:
첨부하신 매뉴얼은 ROS 프로그램 매뉴얼이라 원래 에러코드 관련 내용이 없습니다. 로그상에 올라오는 에러는 ROS에러가 아니라 로봇 제어기에러 입니다. 
로봇 제어기 관련 에러코드 매뉴얼에 대해서는 구매처 및 서비스에 문의하시길 바랍니다.

closed 2020-03-02 by doosan-robotics