DoosanRobotics/doosan-robot #44: python moveit

SeungRyeol, 2020-02-27:
안녕하세요.

moveit 패키지를 이용해서 파이썬 스크립트를 작성하려고 합니다.

먼저 아래와 같은 두산 예제  명령어는 정상적으로 작동합니다. 실행한 명령어는 다음과 같습니다.
```
$ roscore
$ roslaunch dsr_control dsr_moveit.launch model:=m0609 host:=192.168.137.71 mode:=real
$ ROS_NAMESPACE=/dsr01m0609 rosrun moveit_commander moveit_commander_cmdline.py robot_description:=/dsr01m0609/robot_description 

>> use arm 
>> goal0 = [0 0 0 0 0 0]
>> go goal0 
```

파이썬 스크립트에서 작성하기위해 아래 예제를 따라했고, 
[https://ros-planning.github.io/moveit_tutorials/doc/move_group_python_interface/move_group_python_interface_tutorial.html#the-launch-file](url)

아래 코드는 예제 코드 입니다.
[https://github.com/ros-planning/moveit_tutorials/blob/master/doc/move_group_python_interface/scripts/move_group_python_interface_tutorial.py](url)

코드에서 초기 셋팅하는 부분은 다음과 같이 설정했습니다.
```
class MoveGroupPythonIntefaceTutorial(object):
    def __init__(self):
        super(MoveGroupPythonIntefaceTutorial, self).__init__()

        moveit_commander.roscpp_initialize(sys.argv)
        rospy.init_node('move_group_python_interface_tutorial', anonymous=True)

        robot = moveit_commander.RobotCommander("/dsr01m0609/robot_description")

        scene = moveit_commander.PlanningSceneInterface()

        group_name = "arm"
        group = moveit_commander.MoveGroupCommander(group_name, "/dsr01m0609/robot_description")
```
디버깅을 해보니 scene = moveit_commander.PlanningSceneInterface() 부분까지는 정상적으로 작동하지만 그 다음 코드부터 정상적으로 작동하지 않습니다. group 부분에서 아래와 같은 오류 메시지가 뜹니다.

```
Traceback (most recent call last):
  File "/home/shin/catkin_ws/src/dsr_havesting/scripts/moveit.py", line 489, in <module>
    main()
  File "/home/shin/catkin_ws/src/dsr_havesting/scripts/moveit.py", line 439, in main
    tutorial = MoveGroupPythonIntefaceTutorial()
  File "/home/shin/catkin_ws/src/dsr_havesting/scripts/moveit.py", line 105, in __init__
    group = moveit_commander.MoveGroupCommander(group_name, "/dsr01m0609/robot_description")
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/moveit_commander/move_group.py", line 52, in __init__
    self._g = _moveit_move_group_interface.MoveGroupInterface(name, robot_description, ns)
RuntimeError: Unable to connect to move_group action server 'move_group' within allotted time (5s)
```
moveit_commander에서 >> use arm 명령어를 입력했던 것처럼 group_name 변수를 arm으로 입력했는데 오류가 발생합니다.

조언 부탁드립니다. 또한 파이썬 스크립트상에서 사용할 다른 방법이 있다면 부탁드립니다.

감사합니다.

SeungRyeol, 2020-03-09:
혹시 해결까지 시간이 좀 더 걸릴까요?

doosan-robotics, 2020-03-09:
수고가 많으십니다. 
moveit_commander 로 잘 동작하는 것으로 보아 Doosan ROS package 의 문제는 아니라고 예상되며 단순 사용상의 오류라고 생각하고 있습니다. 시간을 할애해서 좀 더 딥하게 살펴보도록 하겠습니다.

SeungRyeol, 2020-03-09:
@doosan-robotics 
넵 매번 신경써주셔서 감사합니다. 빠른 시일 내에 해결되길 바라겠습니다!

doosan-robotics, 2020-03-10:
예상대로 Doosan ROS package 의 문제가 아니고 단순 사용상의 문제였습니다. 

예제 코드(move_group_python_interface_tutorial.py) 에서 단 2곳만 수정하세요. 
 1. group_name = "panda_arm" -> "arm"  으로 수정  (사유: moveIt group name 이 "arm" 입니다.)   
 2. def go_to_joint_state(self): 함수에서 joint_goal 수정 (사유 Frank emika 의 panda 로봇은 7축 입니다. -> 두산 로봇에 맞게 6축으로 수정)   
>>원 소스
    joint_goal[0] = 0
    joint_goal[1] = -pi/4
    joint_goal[2] = 0
    joint_goal[3] = -pi/2
    joint_goal[4] = 0
    joint_goal[5] = pi/3
    joint_goal[6] = 0
>>변경 
    joint_goal[0] = 0
    joint_goal[1] = 0
    joint_goal[2] = pi/2
    joint_goal[3] = 0
    joint_goal[4] = pi/2
    joint_goal[5] = 0

[중요!] 수정 후 실행 시, 네임 스페이스를 변경하시고 rosrun을 하세요(하기 참조)

ROS_NAMESPACE=/dsr01m0609 rosrun {소스위치} {소스명.py}  

ex> ROS_NAMESPACE=/dsr01m0609 rosrun dsr_example_py move_group_python_interface_tutorial.py

SeungRyeol, 2020-03-11:
위에서 말씀해주신대로 1,2번은 수정했었는데, [중요!]로 표시해주신 네임스페이스부분이 문제였네요.

말씀해주신대로 터미널에서 네임스페이스를 추가로 입력하고 rosrun하니 정상적으로 작동합니다.

빠르게 도움주셔서 진심으로 감사드립니다. :)

closed 2020-03-11 by SeungRyeol