DoosanRobotics/doosan-robot #45: Support Melodic

SeungRyeol, 2020-03-11:
안녕하세요.

현재 엔드이펙터단에 3D 비전 카메라와 그립퍼를 부착해 수확로봇을 개발 중 입니다.

AI 모델을 통한 물체인식과 영상처리가 필요하기 때문에 GPU가 포함된 Nvidia Jetson Xavier를 메인 SBC로 선정하였습니다. 

Xavier의 경우 ubuntu 18.04 LTS가 기본 OS이지만, doosan-robot 패키지가 Kinetic 밖에 지원하지 않기 때문에 확정을 보류하고 있습니다.

추후 Melodic 지원 예정이 있는지 궁금합니다.

감사합니다.

doosan-robotics, 2020-03-12:
Melodic 지원은 개발 리소스가 상당히 많이 소모되어 단 시간 안에 지원이 불 가능할 것으로 생각됩니다. 
Xavier 는 16.04 kinetic을 공식적으로 지원하니, 개발하시는데는 문제가 없을 것으로 보입니다.

doosan-robotics, 2020-03-12:
하기 링크와 같은 시스템을 만들려고 하시는 것 같군요.
좋은 결과가 있기를 바랍니다.  (검색어 : Right Hand)
https://www.youtube.com/watch?v=g5i0dsQfPCY

SeungRyeol, 2020-03-12:
[https://devtalk.nvidia.com/default/topic/1042154/jetson-agx-xavier/how-to-install-ubuntu-16-04-in-a-new-nvidia-xavier/](url)

OS 설치를 시도하는 호스트 PC에서는 16.04와 18.04 모두 가능하지만, OS가 깔리는 타겟인 Xavier는 18.04만 지원한다고 합니다.

혹시 Xavier가 ROS Kinetic을 지원한다는 내용이 있나요?

doosan-robotics, 2020-03-12:
자세히 살펴보니, Xavier는 host가 아닌 target으로 돌 때는 18.04만 지원하네요. 
이전 언급을 수정합니다.

SeungRyeol, 2020-03-12:
넵! 그럼 대체 SBC나 다른 방법을 찾아보겠습니다. 답변 감사합니다.

closed 2020-03-12 by SeungRyeol