DoosanRobotics/doosan-robot #46: Moveit Velocity

SeungRyeol, 2020-03-18:
안녕하세요. Moveit 사용 도중 몇가지 질문이 있어 글을 씁니다.

MoveIt을 사용해 모션 플랜을 실행해도 Plan 속도에 비해 매니퓰레이터 동작 속도가 굉장히 느리게 움직입니다.

현재 set_planning_time은 5,
set_max_velocity_scaling_factor은 1(최대)로 설정되어 있습니다.

이동속도를 증가시키기 위해 moveit_config_* 내부에 joint_limits.yaml 파일에서 max_velocity 또는 max_acceleration 등을 바꿔도 되는지 여쭤보고 싶습니다.

또는 속도를 증가시키는 다른 방법이 있나요???

SeungRyeol, 2020-03-18:
터미널 상에서 두산매니퓰레이터의 로그는 다음과 같이 뜹니다.
```
[ INFO] [1584518633.029355864]: Completed trajectory execution with status SUCCEEDED ...
[ INFO] [1584518633.029514695]: Execution completed: SUCCEEDED
[ INFO] [1584518633.427849094]: [callback OnLogAlarm]
[ INFO] [1584518633.427885549]:  level : 1
[ INFO] [1584518633.427893169]:  group : 2
[ INFO] [1584518633.427898536]:  index : 1214
[ INFO] [1584518633.427907209]:  param : [INFO]finalTime(8.345) is adjusted above User Input(3.038)
[ INFO] [1584518633.427914083]:  param : 
[ INFO] [1584518633.427919361]:  param : 
[ INFO] [1584518641.871200489]: On Monitor State
```
3.038초가 인풋인데, FinalTime은 8.345정도 나옵니다.

그리고 팔의 반경을 확트는 동작에서는, 그 동작에 한해서 매우 빠른 속도로 움직입니다.

doosan-robotics, 2020-03-23:
일단 본 이슈는 당사 모션팀과 토의해 본 결과, 분명 개선의 포인트는 있으나, 단 기간에 해결될 이슈가 아님을 알려드립니다. 일단 하기와 같이 2가지 대안을 제시해 드립니다.

1. MoveIt 파라미터를 조절하셔서 최대한 경로가 러프하게 생성되도록 하시길 바랍니다.
  - 로봇 경로가 세밀하게 나올수록 로봇이 이 경로들을 추정해 가기에 부하가 걸리므로 
    로봇 경로가 러프하게 나올수록 유리합니다.

2. 두산 ROS에서 제공하는 모션 서비스(movej, movel, movec...) 를 이용하시길 바랍니다.
  -  손 쉽게 모션 서비스 호출을 통해 두산 로봇을 최적으로 제어할 수 있습니다. 
  - 비전, MoveIt , ROS node 등 여러 알고리즘을 조합하여 나온 최종 타겟 포지션을 
    적절한 모션 서비스 통해서 로봇을 제어하세요. (심플한 방법이지만 강력합니다.)

SeungRyeol, 2020-03-25:
현재 2번의 방법대로 모션 알고리즘을 다시 세워보았습니다.

```
joint = dsr_moveit.degree_to_radian(HAVESTING_HOME) # 타겟 목표에 대한 joint 값 생성
plan = dsr_moveit.group.plan(joint) # moveit을 통한 타겟 포지션 생성
 j_list = [] # movesj를 위한 List 생성
for i in range(0, len(plan.joint_trajectory.points),6):
    j_list.append(posj(dsr_moveit.radian_to_degree(plan.joint_trajectory.points[i].positions))) # 각 위치에 대한 
j_list.append(posj(dsr_moveit.radian_to_degree(plan.joint_trajectory.points[-1].positions))) # 마지막위치에 대한 보장

movesj(j_list, vel=70, acc=70, time = 4) # moveit을 통한 타겟 포지션을 두산 movesj로  이동
```

for i in range(0, len(plan.joint_trajectory.points),6)에서 6간격으로 한 이유는 타겟 목표가 예를 들어 80개가 나왔을 때, 80개의 waypoint를 다 지나가게하면 movesj의 time을 낮게 설정해도 최종 걸린시간이 너무 길게 나와 6개 간격으로 joint 배열을 구성하였습니다.

위와 같이 코드를 구성하는 것이 의도한 것이 맞는지 궁금합니다. 또한 rviz 상에서 모션에 대한 trajectory  plan을 보면 가끔 관절이 확돌아가는 경우가 있는데 파라미터를 설정 오류 일까요? 그 플랜대로 실행하면 관절 제한 값은 벗어나진 않지만 예를 들어 179도에서 -179도로 확 변하는 경우가 존재하긴 합니다. 그렇다보니 매우 빠르게 관절이 꺾여버리고요

doosan-robotics, 2020-03-31:
우선 MoveIt 설정과 파라미터를 잘 설정하시라는 답변을 드립니다. 
상기 코드에서  j_list 값을 프린트 해보시면 MoveIt이 179에서 -179 좌표를 만들고 있는 것으로 예상됩니다. MoveIt 설정이 여의치 않타면  j_list 생성시 필터링을 해서 특이점을 빼버리는 것도 하나의 방법 일듯 싶습니다.

doosan-robotics, 2020-03-31:
지금 개발하시는 어플리케이션에서 꼭  MoveIt으로 경로를 생성하고 중간 경로를 꼭 경우해야만 하나요? 그냥 최종 타켓위치만 movej 로 보내도 될 듯 싶은데요. 
중간경로가 필요없으시면 MoveIt 없이 비전의 보정데이터만 가지고 타켓위치를 계산하고 movej 나 movel 을 움직이시는 것이 나을 듯 싶습니다.

SeungRyeol, 2020-03-31:
최종 경로만 하지 않고 중간경로를 집어넣으려고 한 이유는 다음과 같습니다.

경우1) MoveIt을 통해 경로를 생성한 경로 중, 마지막 좌표만 movej로 이동할 경우 매니퓰레이터와 부착된 모바일 베이스를 고려하지 않고 이동하기 때문에 모바일 베이스(또는 매니퓰레이터가 고정되는 차체)와 충돌하는 경우가 있음

경우2) MoveIt으로 경로생성을 하지 않고 Movel로만  최종 좌표를 만들 경우, 도달할 수 없는 좌표라고 나오는 경우가 있음

위 경우에서 보셨듯이 중간 경로를 집어넣는 가장 결정적인 이유는 차체와의 자체 충돌을 피하기 위해 그렇 습니다. 두산 서비스팀과 통화해 본 결과 차체 사이즈대로 안전 영역 등을 설정해 그 영역에 접근했을 때 멈추게는 할 수 있지만 영역을 고려하여 모션을 계획하지는 않는다는 답변을 받았습니다. 실제로 안전영역이나 작동영역을 설정해보니 설정한 부분을 벗어나면 움직이다가도 로봇 자체가 멈춰버리더군요. 그래서 MoveIt설정상에서 매니퓰레이터가 부착되는 차체를 고려해 자체충돌 영역을 재구성하였습니다.

elton-choi, 2020-03-31:
179 도에서 -179도로 가는 현상은 Inverse Kinematics 이슈입니다. 한가지 task pose에 대해서 여러가지 Inverse Kinematics 해가 있을 수 있습니다. 따라서, 연속적인 포인트를 이어줄 때, 이전 자세와 다음 자세의 Joint space  에서 거리를 고려하지 않으면, task 상에서는 planning한 자세로 가지만 Joint에서는 갑작스러운 회전이 들어갈 수 있습니다. Moveit에서 일반적인 이슈로 보이는데 관련 이슈의 해결법을 몇 가지 시도해보시면 좋을 것 같습니다. 
[https://github.com/ros-planning/moveit/issues/750](https://github.com/ros-planning/moveit/issues/750)
[https://answers.ros.org/question/265699/strange-tool-flip-in-descartes/](https://answers.ros.org/question/265699/strange-tool-flip-in-descartes/)

elton-choi, 2020-03-31:
Track IK 가 solve type 에서 distance를 고려할 수 있는 것으로 보입니다. 한번 시도해보셔도 좋을 것 같습니다.
[https://ros-planning.github.io/moveit_tutorials/doc/trac_ik/trac_ik_tutorial.html](https://ros-planning.github.io/moveit_tutorials/doc/trac_ik/trac_ik_tutorial.html)

SeungRyeol, 2020-04-01:
@elton-choi 
다음과 같이 kinematics를 설정하였지만 같은 결과가 나타납니다. 제가 셋팅을 잘못했을까요?
```
# moveit_config_m0609/config/kinematics.yaml
arm:
  kinematics_solver: trac_ik_kinematics_plugin/TRAC_IKKinematicsPlugin
  kinematics_solver_timeout: 0.05
  solve_type: Distance
```

```
# kinematics 관련 노드 실행 정보
 * /dsr01m0609/robot_description_kinematics/arm/kinematics_solver: trac_ik_kinematic...
 * /dsr01m0609/robot_description_kinematics/arm/kinematics_solver_timeout: 0.05
 * /dsr01m0609/robot_description_kinematics/arm/solve_type: Distance
```

아래는 trac IK 동작화면 입니다.
https://drive.google.com/file/d/17bHockbU_NTVY1eKHiM8p02JUAzqvy-S/view?usp=sharing

SeungRyeol, 2020-04-01:
다른 solve_type인 Manipulation1, Manipulation2 등을 사용해 보았는데 이 두가지 경우는 위 영상과 같은 상황에 대해 플랜을 생성해내지 못합니다. (출력 plan = 제자리)

elton-choi, 2020-04-02:
올려 주신 모션을 보니까, ik만의 문제는 아니고 계속 앞으로 뻗으면 엘보우가 탁자에 부딪힐 것 같으니까 planner가 숄더를 반대로 틀고 엘보우를 반대로 꺾은 것 같네요. 처음 자세를 마지막 자세처럼 숄더는 반대로 엘보우도 위로 가게 하시면 꺾일 일이 없을 것 같구요.

처음 자세를 조정하시는 게 불가능 하시면 movesj 보다는 movej를 블렌딩을 줘서 쓰시면 더 나을 것 같습니다. 이유는 movesj(vel=70, acc=70, time=4)으로 쓰시면 vel, acc, time이 한꺼번에 들어올 때 vel, acc는 무시되고 time을 기준으로 모션을 만들게 되구요. 4초가 안지켜지는 이유는 전체 모션 중에 로봇 최대 속도를 넘는 구간이 있어서 그러지 않도록 자동적으로 4초를 늘려주기 때문입니다. 하지만 movesj(vel=70, acc=70)으로 주셔도 어찌됐든 꺾이는 곳이 최대 속도가 되기 때문에 꺾이는 곳을 vel=70으로 만들면서 전체 구간 속도를 낮춰버리는 문제가 생깁니다. 다시 처음에 질문하셨던 원점으로 돌아가게 되는거죠.

제 생각에는 2가지 방법이 있는데,
1. movej 사용
for i in range(0,n)
    movej(j_list[i], v=70, a=70, r=50)

movej를 쓰면 전체 구간을 보지 않고 세그먼트마다의 속도대로 따로 움직여주기 때문에 전 구간
조인트에서는 일정한 속도로 움직일 수 있구요. r은 각 구간의 가감속이 겹쳐지는 거리로 주시면
끊어지는 느낌이 덜할 것 같습니다. 하지만, r이 현재 위치와 타겟 위치의 1/2 보다 작게 넣어주셔야 하기 때문에 그 부분은 fkin 함수를 이용해서 각 포즈 간 차이를 계산하신 다음 빼서 1/2보다 작게 넣어주시도록 자동화하셔도 될 것 같습니다.

SeungRyeol, 2020-04-02:
@elton-choi 
답변 감사합니다. 시작 위치를 적절히 배치해 테스트를 해보니 더 좋은 결과가 나왔습니다. 모션은 movej로 했었고요. 그리고 블렌딩도 한 번 적용해보려고 하는데 말씀해주신 fkin 함수는 파이썬 스크립트 상에서는 없다고 나오는데 DRL에서만 사용가능한 거 아닌지 여쭤보고 싶습니다.

elton-choi, 2020-04-03:
그럴 수도 있겠네요. @doosan-robotics  확인 부탁드립니다.
중간에 약간 끊기는 느낌이 상관없으시면 blending은 r=1만 주셔도 되구요.
planning할 때 각 포인트의 distance를 설정한 게 있으시다면 그 절반 값으로 constant하게 주셔도 됩니다.

doosan-robotics, 2020-04-07:
답변이 늦었습니다. 
fkin 은 현재 서비스가 없습니다. 이제 곧 배포될 버전에는 포함됩니다. fkin 이외에도 힘제어 관련 서비스들을 제공할 예정입니다. 조금만 기다려 주시면 감사하겠습니다.

SeungRyeol, 2020-04-09:
넵. 도움을 주신 분들 모두 감사합니다.

closed 2020-04-09 by SeungRyeol

adrianwslee, 2020-12-22:
@doosan-robotics 

> 일단 본 이슈는 당사 모션팀과 토의해 본 결과, 분명 개선의 포인트는 있으나, 단 기간에 해결될 이슈가 아님을 알려드립니다. 일단 하기와 같이 2가지 대안을 제시해 드립니다.

라고 하셨는데 혹시 타겟 시간이나 개선점 업데이트 된 게 있을까요?

elton-choi, 2020-12-23:
@doosan-robotics 
안녕하세요, 당사 모션팀인데요.
해당 이슈 개선은 내년 6월까지로 잡혀있습니다. (버전 발행 계획에 따라 좀 더 앞당겨 질 수도 있습니다.)
개선 부분은 movesj, amovesj 개선과 신규 servoj, speedj 등 명령어를 만들어서 joint trajectory controller에서 생성되는
명령을 속도 저하 없이 따라갈 수 있게 할 계획입니다.

nicola-sysdesign, 2021-12-07:
> @mpavezb @adrianwslee movesj/amovesj is improved at v2.8. Even if it is not complete solution, we removed conservative acceleration limit hidden in Control SW. You can use movesj/amovesj with similar speed of blended movej function. You can try this v2.8 or later.

Hi, I'm facing the same issue with an H2017 during execution of some trajectories generated by MoveIt.
```
[ INFO] [1638884353.692610061]: Completed trajectory execution with status SUCCEEDED ...
[ INFO] [1638884353.692837783]: Execution completed: SUCCEEDED
[ INFO] [1638884353.883985809]: [callback OnLogAlarm]
[ INFO] [1638884353.884075049]:  level : 1
[ INFO] [1638884353.884124646]:  group : 2
[ INFO] [1638884353.884181790]:  index : 1214
[ INFO] [1638884353.884236272]:  param : [INFO]finalTime(3.817) is adjusted above User Input(1.694)
[ INFO] [1638884353.884299196]:  param : 
[ INFO] [1638884353.884538528]:  param : 
[ INFO] [1638884357.784632193]: On Monitor State
```
Sorry for my ignorance, but where can I check If I'm using the software version you are talking about?

InigoMoreno, 2022-03-10:
I am running version 2.8 of the robot software and I still see this problem.