DoosanRobotics/doosan-robot #50: ROS melodic support

bgyoo91, 2020-04-20:
안녕하세요.

점점 로봇 관련 센서들이 18.04 이상 지원하고 있는데,
ROS melodic 관련하여 doosan robotics 에서 향후 계획이 있는지 문의 드립니다.
혹은 이전에 melodic 을 지원했을때의 코드를 받을 수 있을까요?

doosan-robotics, 2020-04-22:
곧 배포 될 doosan ROS package 업데이트 이 후, 
ROS melodic 지원 건에 대해 심도 있게 검토하겠습니다. 
참고로 이전에 melodic 을 지원했을때의 코드는 문제가 있어 공유가 의미가 없습니다.

bgyoo91, 2020-04-22:
감사합니다:)

simonschmeisser, 2020-04-24:
Could you please give an estimate when soon will be? We currently have a Doosan at our company to integrate it into our bin-picking software but we only have access to it for two weeks (till 4th of may), will the release be before that date?

doosan-robotics, 2020-04-29:
두산  ROS 패키지를 업데이트 받으시길 바랍니다. melodic에서 사용 가능 합니다.

doosan-robotics, 2020-04-29:
Please update the Doosan ROS package. Available in melodic.

axelschroth, 2020-04-29:
Thank you very much for this quick response. We will test the package within the next few days.

axelschroth, 2020-05-04:
First test run completed with following results (using m1013):

- Executing trajectories using dsr_control / MoveIt works under ros melodic
- ROS services: service calls sometimes cause exceptions. Exception messages are not printed to console
- I/Os:
  - Writing outputs works
  - Reading state of outputs not implemented (but one can use the `state` topic)
  - Reading inputs does not work. I was getting exceptions when calling `io/get_digital_input`. Surprisingly I got value `0` when reading `io/get_tool_digital_input "index: 0"` which should not exist
- Topic `state` is useful, but does not follow RobotStatus message of ros industrial (see http://docs.ros.org/melodic/api/industrial_msgs/html/msg/RobotStatus.html). Are you going to implement that topic publisher as well?
- Running DRL code: I tried to execute some simple commands using `drl/drl_start` service. Every try I made resulted in an exception. E.g. I tried via command line:

  ```
  rosservice call /dsr01m1013/drl/drl_start "robotSystem: 0
  code: 'set_digital_output(6, ON)'" 
  ```
  
Of course I did not test every service / functionality you currently provide, but I think these points may be helpful. I am not sure, which issues are caused by the melodic version, if some are not a follow up issue may be appropriate.

It would be great, if you could respond to my problems, so we can continue with our integration shortly. Thanks!

kabdol2, 2020-05-05:
It's a long holiday in Korea right now and today is the last day.(Not because of COVID-19, just a public holiday.)
When our ROS team returns to the company, we will find answers to your questions.
Thank you for your interest in the Doosan ROS package, and I hope you can use this package to make great applications.
If you have the chance, I hope you will show and share your project to many people.
That would be a lot of encouragement to our developers, including me.

doosan-robotics, 2020-05-06:
@axelschroth 
It is the answer to your question or my question.

- Executing trajectories using dsr_control / MoveIt works under ros melodic

- ROS services: service calls sometimes cause exceptions. Exception messages are not printed to console
  **>>>Specifically, in what service does the error occur? Is it blocked without outputting an error message?**

- I/Os:
  - Writing outputs works
  - Reading state of outputs not implemented (but one can use the `state` topic)
  - Reading inputs does not work. I was getting exceptions when calling `io/get_digital_input`. Surprisingly I got value `0` when reading `io/get_tool_digital_input "index: 0"` which should not exist
  **>>>I tested it and it works fine. For reference, in the example, the robot status message is updated every 1 second.**
  **>>>The return value of get_digital_input is given as bool type as True(1) or False(0).**
  **>>>Error handling is not performed when the index range is exceeded. We will handle the error in the next update.**

- Topic `state` is useful, but does not follow RobotStatus message of ros industrial (see http://docs.ros.org/melodic/api/industrial_msgs/html/msg/RobotStatus.html). Are you going to implement that topic publisher as well?
  **>>>As many robot companies (KUKA, UR, Franka, and Rethink) have been benchmarked, most of them do not follow the Robot Status message of ros-industrial.**

- Running DRL code: I tried to execute some simple commands using `drl/drl_start` service. Every try I made resulted in an exception. E.g. I tried via command line:

  ```
  rosservice call /dsr01m1013/drl/drl_start "robotSystem: 0
  code: 'set_digital_output(6, ON)'" 
  ```
  **>>>It works fine in our environment. Please give us in detail about your TEST environment.**

- I had to remove metapackage `<build_depend>industrial_core</build_depend>` in `doosan_robotics/package.xml`, otherwise catkin wouldn't build without an error (industrial_core only has run_dependencies by the way)

  **>>>In our environment (x86, ubuntu18.04), it builds fine without modification.
Please give us in detail about your TEST environment.**

- Xacros: as described [here](http://wiki.ros.org/Industrial/Tutorials/Create%20a%20URDF%20for%20an%20Industrial%20Robot) in Section "2.2.1 Additional/Standard Frames", some conventions are defined in regard to link naming (i.e. of `base_link` which is called `base_0` in your xacros). An optional argument named `prefix` is used in many ros industrial robots as well. I am going to submit a Pull Request for that.
  **>>>I am looking forward to your Pull Request.!**

Of course I did not test every service / functionality you currently provide, but I think these points may be helpful. I am not sure, which issues are caused by the melodic version, if some are not a follow up issue may be appropriate.

It would be great, if you could respond to my problems, so we can continue with our integration shortly. Thanks!

doosan-robotics, 2020-05-06:
@bgyoo91 
bgyoo91님 melodic 환경에서 TEST 해보셨나요? 문제점이 있으면 여기에 남겨 주시면 감사하겠습니다.
bgyoo91, have you tried to test in a melodic environment? If you have any problems, please leave it here.

bgyoo91, 2020-05-06:
@doosan-robotics 빠른 대응에 감사합니다. 되도록 이번주 안에 기존의 저희 어플리케이션이 melodic 버전에 잘 동작하는지 확인해볼 예정입니다.

axelschroth, 2020-05-08:
Thank you very much for the quick response.

Firstly my test setup is Kubuntu 18.04, x86_64 platform running ROS melodic. Robot: Doosan 2.5 firmware, M1013.

> * ROS services: service calls sometimes cause exceptions. Exception messages are not printed to console
>   **>>>Specifically, in what service does the error occur? Is it blocked without outputting an error message?**

E.g. service `io/get_digital_input` throws an exception (if index != 0). Service call does not get blocked, command line just outputs that an exception occurs (without any further information).

> * I/Os:
>   * Reading state of outputs not implemented (but one can use the `state` topic)

Are you planning to add a service for that purpose, which reads current output state from the doosan controller?

> * Reading inputs does not work. I was getting exceptions when calling `io/get_digital_input`. Surprisingly I got value `0` when reading `io/get_tool_digital_input "index: 0"` which should not [...]

Maybe my issues are related to my test setup (see above, maybe 64bit version causes problems)?

> * Topic `state` is useful, but does not follow RobotStatus message of ros industrial (see http://docs.ros.org/melodic/api/industrial_msgs/html/msg/RobotStatus.html). Are you going to implement that [...]

Some do so, some not. Current version of Fanuc and UR ros driver supports it. Regarding Kuka and Mitsubishi i.e., we added this message type by ourselves.

> * Running DRL code: I tried to execute some simple commands using `drl/drl_start` service. Every try I made resulted in an exception. E.g. I tried via command line:
>   ```
>   rosservice call [...]

Maybe this also is related to my test setup..? Same problem with service calls as stated above with reading I/Os..

elton-choi, 2020-05-09:
@doosan-robotics 
사소한 부분이지만, README.md의 guide에 kinetic이라고 명시된 부분에 대해 $ROS_DISTRO 로 바꿔서 범용적으로 사용할 수 있게 하던지, kinetic-devel, melodic-devel에 따라서 README.md 를 다르게 적용하는 게 좋을 것 같습니다.

doosan-robotics, 2020-05-11:
@elton-choi 
귀하의 요구 사항을 반영하였습니다.
We reflected your needs.

doosan-robotics, 2020-05-11:
@axelschroth 
There is a problem with the drl service call. Do not make drl service calls. If the drl service call is made even once, the ros service does not operate normally.

doosan-robotics, 2020-05-11:
@axelschroth 
I tested it on ubuntu x86(64bit) environment today.
On the melodic_devel branch, it should build normally without any modifications. The only difference between your environment and my environment is the difference between kbuntu and ununtu. It's hard to see the difference between kbuntu and ununtu causing your problem. 
Please check your environment.
If the problem is still not resolved, please leave a log when you run "roslaunch dsr_control dsr_moveit.launch model: = m1013"

doosan-robotics, 2020-05-12:
I tested it on **kbuntu 18.04**. It worked fine as expected.

axelschroth, 2020-05-12:
Okay, that seems strange. I will do further testing and investigate my logs as soon as I have access to the robot again.
You tested I/O service calls I guess?

> There is a problem with the drl service call. Do not make drl service calls. If the drl service call is made even once, the ros service does not operate normally.

Are you going to fix that?

doosan-robotics, 2020-05-15:
Minor updates were implemented. Refer to your development,

1. Insert return value success into all * .srv files
2. Add new service
  -GetCtlBoxDigitalOutput.srv
  -GetToolDigitalOutput.srv
3. New Service Function
  -get_digital_output (index)
  -get_tool_digital_output (index)

axelschroth, 2020-06-02:
> I tested it on **kbuntu 18.04**. It worked fine as expected.

Using the melodic-devel branch and following catkin version:

```
$ catkin --version
catkin_tools 0.4.5 (C) 2014-2020 Open Source Robotics Foundation
catkin_tools is released under the Apache License, Version 2.0 
(http://www.apache.org/licenses/LICENSE-2.0)
---
Using Python 2.7.17 (default, Apr 15 2020, 17:20:14) [GCC 7.5.0]
```

Building the doosan_robotics package results in following output:

```
$ catkin build doosan_robotics
...  
Abandoned <<< doosan_robotics                                 [ Depends on unknown jobs: industrial_core ]
...
```

As stated in https://www.ros.org/reps/rep-0127.html#metapackage, metapackages are only allowed to be referenced in other metapackages using <run_depend>.

doosan-robotics, 2020-06-03:
>We performed catkin build on __kubuntu 18.04__ melodic, and the following result was output.
[image]

> As in the above issue, _$ catkin build doosan-robotics_ command was used, but the **Abandon** message could not be confirmed.

axelschroth, 2020-06-03:
After some further testing, I'd like to share following results and suggestions:

- Service calls related to I/O writing and reading (inclusive new ones for getting output values) do work properly with most recent version
- Service call `drl_start` works now without an exception.
  When calling
  ```
  rosservice call /dsr01m1013/drl/drl_start "robotSystem: 0
  code: 'set_digital_output(0, 1)'" 
  success: True
  ```

  I am not sure if success: True is expected behaviour since index 0 does not exist and causes an error on robot controller.
- No abrupt stops so far during motion

_Limitations/Questions:_

- Are you planning to remove the limit of 100 trajectory points?
- Are you planning to consider time parametrization of trajectories? Currently just total time of trajectory is passed to robot, but not time estimations between trajectory points. Changes in motion speed within the trajectory is not possible this way.

doosan-robotics, 2020-06-24:
I will close this issue because it works fine in the melodic version.
If there are any other issues, I would be grateful if I could create a new one to suit the subject.

closed 2020-06-24 by doosan-robotics