DoosanRobotics/doosan-robot #51: Missing Robot Feedback, incomplete implementation with arm-controller (Read/Write-function)

sRogge, 2020-06-03:
https://github.com/doosan-robotics/doosan-robot/blob/d0d6f1b3ffe30896fa11adad83d604673e578ea9/dsr_control/src/dsr_hw_interface.cpp#L987

Dear Doosan Team,

We have tried to use the Doosan robot in conjunction with moveIT and it was important to us that we could stop the robot movement at any time. This is what the ROS template with the arm controller provides for. Unfortunately this does not work with the Doosan driver because the read/write functions are not fully implemented. 

A very detailed error documentation including pictures and sources can be found in the attached pdf in German and English.

Are there any ambitions to fix this bug in the near future?

Thanks and best regards
Simon

[Doosan_Driver_v2.pdf](https://github.com/doosan-robotics/doosan-robot/files/4722127/Doosan_Driver_v2.pdf)
[Doosan_Treiber_v2.pdf](https://github.com/doosan-robotics/doosan-robot/files/4722129/Doosan_Treiber_v2.pdf)

gavanderhoorn, 2020-06-16:
@sRogge: this will be off-topic here, but I've just read the `.pdf` you've attached and I'd like to warn you about using `ur_driver` for anything but historical curiosity.

The code in there should not be used for any purpose, as it's fully deprecated.

I'm not entirely sure what you mean by "the path is also tapped" (on page 5) and "Universal Robots does the same thing, for example" (also on page 5), but I believe there is some misunderstanding there.

If you have any questions, please post an issue on either the `ros-industrial/universal_robot` tracker, or the one of `UniversalRobots/Universal_Robots_ROS_Driver` and we can clear them up for you.

doosan-robotics, 2020-06-24:
As a result of discussions with other related teams on this issue, unfortunately, it is difficult to input resources for this development for the time being.

closed 2020-07-02 by doosan-robotics

gavanderhoorn, 2020-07-02:
@doosan-robotics: by closing this issue, is that a confirmation that at least for the time being you will not be addressing the issues raised by @sRogge?