DoosanRobotics/doosan-robot #52: Error launching $ roslaunch dsr_launcher single_robot_rviz.launch model:=m1013 gripper:=robotiq_2f

wilderdyd, 2020-06-21:
Hi, Hi, I launched **$ roslaunch dsr_launcher single_robot_rviz.launch model:=m1013 gripper:=robotiq_2f** and when the robot is displayed in gazebo the gripper(robotiq 2f) is shaking like a Parkinson's disease. 

[image]

Can you help me with this issue and tell me how I can solve it?

doosan-robotics, 2020-06-24:
We found a problem with the __common__ pacakge tree structure. Due to this, the plugin for the robotiq gripper model was not built normally.
Fixed CMakeList.txt file of __common__ package, and the problem does not occur anymore.

closed 2020-07-02 by doosan-robotics

SANGEERTH-P, 2024-07-01:
The problem is still persistent sir. The gripper is dancing randomly. The rviz is also not getting loaded properly
[image]
Can you please help with this issue?