DoosanRobotics/doosan-robot #55: Move it commander example not working properly.

Ryudas, 2020-08-11:
I have setup everything for the doosan m1013. and tried to launch  moveit commander from the manual. I launched like so:
roslaunch dsr_control dsr_moveit.launch model:=m1013 mode:=virtual

however i get the following errors when launching this:
ERROR: cannot launch node of type [dsr_control/dsr_control_node]: can't locate node [dsr_control_node] in package [dsr_control]

[ERROR] [1597155302.606933879]: Action client not connected: dsr_joint_trajectory_controller/follow_joint_trajectory

installing the workspace gave me no errors.

doosan-robotics, 2020-08-12:
The command'roslaunch dsr_control dsr_moveit.launch model:=m1013 mode:=virtual' runs without problems.
Can you capture the error screen?

[image]

Ryudas, 2020-08-12:
[image]
dsr_control is properly installed and i went through all the points of the installation process. 
[image]

Ryudas, 2020-08-12:
after some searching, I used catkin_make again only on dsr_control, and that has fixed it. not sure why I would have to use this again however, as all the other packages worked fine

closed 2020-08-12 by Ryudas