DoosanRobotics/doosan-robot #57: Move it + DSR _control does not work on real robot

Ryudas, 2020-09-02:
Hello, I am trying to deploy the examples in the manual for DSR control + moveit but though dsr control launches, reviz is stuck in the initialization screen and never begins (running on kinetic)  - I do get control from the pendant  (it unlocks the brakes) :    
[image]
[image]
[image]

doosan-robotics, 2020-09-03:
Is running rviz standalone(rosrun rviz rviz) okay?
If there is a problem, it could be a graphics driver problem.
Please refer to the this [link](https://answers.ros.org/question/265030/moveit-demolaunch-rviz-initializing-forever/).

Ryudas, 2020-09-03:
> Is running rviz standalone(rosrun rviz rviz) okay?
> If there is a problem, it could be a graphics driver problem.
> Please refer to the this [link](https://answers.ros.org/question/265030/moveit-demolaunch-rviz-initializing-forever/).

Standalone rviz has the same issue.  I want to point out, using the **virtual** launch, everything is fine, this only happens when trying with **mode:=real**.

doosan-robotics, 2020-09-09:
Is rviz run normally in virtual  mode? If rviz does not run independently, it may be a problem with the execution environment (graphics driver, display environment, etc.).

Ryudas, 2020-09-09:
rviz works absolutely fine in virtual mode. If i try exactly the same roslaunch, (but with mode:=real and the host, then rviz is stuck forever in the screenshots I've sent.

Ryudas, 2020-09-15:
I've manaeged to fix the error by reinstalling the full environment again and the driver.  I am not sure what is the exact cause, especially as with simulation it worked fine.

closed 2020-09-15 by Ryudas