DoosanRobotics/doosan-robot #59: Compliance mode

YWJae, 2020-09-16:
안녕하세요,

Compliance mode를 사용하기 위해 'set_stiffnessx' 함수를 사용하고 있는데요.

로봇 end-effector의 무게(약 2kg) 때문에 관절에 걸리는 load 때문인지 특정 경우에 'set_stiffnessx' 함수에서 코드가 멈추는 문제가 계속 발생해 compliance 모드를 활용한 개발에 어려움을 겪고있습니다.

관련 service를 call 해도 return이 되지 않는 것을 봐서 두산로봇 컨트롤러 내부적으로 compliance 모드 관련 핸들링 알고리즘이 있는 것 같은데, 혹시 해결 방안이 있는지 문의드립니다.

doosan-robotics, 2020-09-21:
원인 분석을 위하여 사용 프로그램의 script 공유 부탁드리겠습니다.

elton-choi, 2020-09-26:
일단은 Compliance mode를 사용하려면
task_compliance_ctrl() 함수를 먼저 호출해야 합니다.
set_stiffnessx는 task_compliance_ctrl 함수 다음에 stiffness를 변경할 때 사용하는 함수입니다.
task_compliance_ctrl() 함수 예제를 참조하세요.
그리고, 2kg의 툴은 다셨다면 tool 무게 설정을 해주세요. 무게 설정을 하지 않으면 2kg의 툴을 외력으로 인식해서
컴플라이언스 컨트롤 진입이 안될 수 있습니다.(로봇 입장에서는 외력이 있는 상태를 위험한 상태로 인식합니다.)

closed 2020-11-03 by doosan-robotics

YWJae, 2020-11-03:
아, 답변 확인을 지금 하였네요..

말씀하신 것 처럼 tool의 무게를 설정해서 문제를 해결하였습니다.

답변 진심으로 감사드립니다.

윤원재 올림