DoosanRobotics/doosan-robot #62: Question: frequencies of external interfaces to Doosan Controller

rtonnaer, 2020-11-18:
This is more a question then an issue.

In the [dsr_hw_interface](https://github.com/doosan-robotics/doosan-robot/blob/master/dsr_control/src/dsr_hw_interface.cpp) I see mention of multiple freqiemcoes

- [`DSR_CTL_PUB_RATE: 100 Hz`](https://github.com/doosan-robotics/doosan-robot/blob/8a19e3e8fad021b34e0cf31e7a1833dcd01caba7/dsr_control/src/dsr_hw_interface.cpp#L30)
- [in `DRHWInterface::OnMonitoringDataExC`: 10 Hz](https://github.com/doosan-robotics/doosan-robot/blob/8a19e3e8fad021b34e0cf31e7a1833dcd01caba7/dsr_control/src/dsr_hw_interface.cpp#L208)

The Robot Controller has many different interface options:
- Serial
- TCP
- Modbus
- EthernetIP
- Profinet

Is anyone from Doosan able to confirm (for each interface) what the maximum achievable frequencies are for **motion** and **I/O**?

doosan-robotics, 2020-11-20:
@rtonnaer 
Monitoring data is refreshed every 100ms to prevent CPU overload.

Serial : It depends on the set baudrate.
TCP : It may be different depending on the Ethernet environment.
Modbus : It may be different depending on the Ethernet environment.
EthernetIP : It may be different depending on the Ethernet environment.
Profinet : It may be different depending on the Ethernet environment.

farshbafdoustar, 2021-03-05:
@doosan-robotics 
Is there anyway to communicate with the controller in realtime? with fixed time intervals ? sth like KUKA RSI 4ms (joint interface) and 12ms(Cartesian interface) ?
Is it possible to set position or velocity setpoints for the joints over Profinet?

doosan-robotics, 2021-03-11:
It will be supported in the future.
Real-time communication is currently difficult.
Also, there is no plan to support Profinet on ROS.

doosan-robotics, 2021-04-08:
We are currently developing a low level interface with 1ms.

RemiFabre, 2021-04-08:
Good to hear, this is really needed for us. Any approximate date for this?

doosan-robotics, 2021-04-12:
It is still in the early stages of development, so it is difficult to answer a specific schedule. After some progress, We will reveal the approximate schedule.

pucciland95, 2024-01-24:
@doosan-robotics, any news about the real time interface?

Cheers