DoosanRobotics/doosan-robot #66: M1013모델 정확도 질문

VIduhyeon, 2021-01-13:
안녕하세요? m1013모델을 사용하고 있습니다.

moveit plan을 통해 나온 각도값을 movej로 실행시킬시, tf 상으로는 동일한 x,y,z목표값이 나오지만 실제 로봇이 위치한곳은 동일하지가 않습니다.

목표지점 좌표( x = -0.052, y = 0.236, z = 1.316 )에 대한 IK의 최종지점 각도는 다음과 같이 3가지가 나왔습니다.
1. [-69.292, -36.469, 49.835, -0.015, -13.352, 69.262]
2. [-265.832, -10.620, 49.810, 179.937, 39.155, -274.084]
3. [290.717, -36.453, 49.808, 179.868, 13.325, 249.379]

해당 각도들로 movej을 실행시켰을 경우, 서로 다른 위치에 로봇 팔이 위치하게 되며 약 4cm의 오차범위를 가지고 있습니다.
이 정도의 오차범위가 합당한 것인지, 제가 어떤 부

분을 건들이면 좋을지 조언 주시면 감사하겠습니다.

아래의 영상은 위 3가지 각도에 대한 로봇팔의 실제 움직임과 tf_echo를 통해 나오는 좌표값을 나타낸 것입니다.

영상 1 - 3가지 각도에 대한 로봇팔 위치( 바닥의 레이저포인터를 참고해 주세요)
-> 3가지 각도에 대한 위치 모두 다른 것을 확인할 수 있습니다.

https://user-images.githubusercontent.com/40200333/104422656-687c7200-55c0-11eb-899d-d2cea95d9e71.mp4

영상 2 - 3가지 각도에 대한 tf_echo결과( world <-> link6)  
-> 3가지 각도에 대한 tf변환 값은 동일하게 나오는것을 확인할 수 있습니다. (x = -0.052, y = 0.236, z = 1.316)

https://user-images.githubusercontent.com/40200333/104424588-e9d50400-55c2-11eb-8d45-1ff6a3c06e4f.mp4

감사합니다.

doosan-robotics, 2021-01-14:
하기와 같이 답변드립니다.

1. 동일한 TASK 좌표에서의 서로다른 JOINT configuration으로 움직일 때 Flange의 위치오차는 정확도에 관한 것으로 로봇의 정확도는 당사가 보장하지 않습니다. (고객에게 제시되지 않는 값임을 양지해 주십시오)

2. 로봇의 정확도 오차가 있을 수 있으나 본 테스트에서 4Cm의 큰 오차가 나 보이는 것은 로봇Flange의 오차를 재는 것이 아니라 레이저를 이용하여 Flange에서 먼 거리에 있는 바닥에서 측정하기 때문으로 보입니다. (레이저포인터로 먼거리를 주시하고 있을 때 손목의 작은 움직임으로도 레이저가 보이는 포인트는 크게 움직인다는 것을 생각하면 됩니다.)

3. 검증을 위해서 table등을 놓아 바닥보다 가까운 위치에서 레이저의 탄착점을 측정해보십시오. 기존의 오차보다 줄어들 것입니다.

elton-choi, 2021-01-14:
안녕하세요, 추가적으로 조언을 드리자면,
tf에서 나오는 것은 툴이 입력하신 tcp position, orientation에 정확하게 설치된다는 가정하에 동일한 좌표가 나오는 것이구요. 실제로는 영상으로 유추했을 때 툴의 orientation과 위치가 정확하게 가공하여 붙이신 것이 아니라서 거기서 생긴 위치/각도 오차가 실제 레이저 포인터로 더 멀리 갔을 때 반영되는 것입니다.
일단은 orientation을 맞추는 것이 더 힘들 수 있기 때문에 레이저를 쏘는 툴 끝단이 아니고, 레이저가 지면에 가리켜야 하는 가상의 툴 끝의 position을 맞춘다고 생각하고, 그 position을 측정장치를 이용하여 calibration하거나 그게 여의치 않으시면 튜닝을 통해 찾아서 TCP로 입력해주셔야 합니다.
두산 로보틱스 TCP 설정 메뉴에 TCP Calibration Menu가 있는데, TCP 끝단을 정확하게 동일한 점으로 하되 orientation을 다르게 네 점을 티칭하면 TCP position을 구해줍니다. (orientation은 구해주지 않습니다.)
이것을 이용하면 Laser 포인터로 바닥의 한 점을 가리키되, orientation을 다르게 네 점을 티칭해서 가상의 TCP position을 계산할 수 있는데, 이 때 중요한 것은 실제 툴에서 가상의 tcp 끝점까지 동일한 거리가 유지되도록 티칭하셔야 합니다. 티칭이 부정확할 수록 네 점이 아니라 많은 점이 필요한데, 그럴 경우 두산 로봇은 네 점까지만 Calibration을 제공하므로 직접 tool calibration을 하셔야 합니다.

VIduhyeon, 2021-01-15:
@elton-choi @doosan-robotics 
두산 로보틱스 TCP 설정 메뉴에서  TCP Calibration Menu라고 하셨는데 혹시 TP 내에서 어디로 가야하는 지 알 수 있을까요? 혹시 User Coordinates에서 apply teaching points 부분을 말씀하시는 건가요? 

> 레이저가 지면에 가리켜야 하는 가상의 툴 끝의 position을 맞춘다고 생각하고
이 부분이 이해가 안가서 그러는데 추가 설명 부탁드립니다.

VIduhyeon, 2021-01-19:
@elton-choi @doosan-robotics 
안녕하세요.

> 로봇의 정확도 오차가 있을 수 있으나 본 테스트에서 4Cm의 큰 오차가 나 보이는 것은 로봇Flange의 오차를 재는 것이 아니라 레이저를 이용하여 Flange에서 먼 거리에 있는 바닥에서 측정하기 때문으로 보입니다. (레이저포인터로 먼거리를 주시하고 있을 때 손목의 작은 움직임으로도 레이저가 보이는 포인트는 크게 움직인다는 것을 생각하면 됩니다.)

> 검증을 위해서 table등을 놓아 바닥보다 가까운 위치에서 레이저의 탄착점을 측정해보십시오. 기존의 오차보다 줄어들 것입니다.

말씀하신대로 위의 원인이 4cm오차의 원인이였습니다. 

그래서 실험 방법을 바꾸어 테스트를 진행해 봤습니다.
먼저 moveit에서 나오는 IK해가 아닌 티치펜던트에 x : -54, y : 236, z : 1316, Rx : 0, Ry : 0, Rz : 0의 값을 주고 로봇 팔의 시작 각도는 서로 다르게 두었습니다.(같은 목표지점이라도 다른 Joint값이 나오도록)

1. 레이저 포인터를 TCP중앙에 위치하게 장착
-> 해당 테스트 조건으로 실험해 보았을때 TASK좌표계 기준으로 서로다른 Joint Configuration으로 동작시켰을때 y축으로 약 0.7cm의 오차가 발생하였습니다.

2. 레이저 수평계를 하나의 Joint Configuration에 맞추고 실험을 진행해 봤습니다.
해당 사진은 같은 티치펜던트에서 같은 TASK좌표를 입력하였을때 2가지 경우 입니다.
사진상으로 확인 되듯 약 0.7cm의 오차가 동일하게 발생함을 확인하였습니다.

[image]
< TASK상 같은 좌표에 대한 Joint Configuration 1>
[image]
< TASK상 같은 좌표에 대한 Joint Configuration 2>

해당 위치로 로봇팔이 이동하였을때 티치 펜던트상 위치 오차는 1mm이내 인것을 확인하였습니다.
그러나 실제 로봇팔의 움직임을 보았을때는 0.7cm의 오차가 발생하는데, 의심이 가는것은 로봇모델의 수치가 부정확한것으로 생각됩니다.

해당 오차를 줄이기 위한 기술적 지원을 받을 수 있는지, 아니면 이정도 오차범위는 정상적인것인지 궁금합니다.

감사합니다.

elton-choi, 2021-01-20:
@VIduhyeon 
안녕하세요, 일단 로봇의 반복정밀도는 스펙에 있으나 정확도에 대해서는 스펙으로 보장하지는 않아서 공식 지원을 받기는 어려우실 것 같습니다. 물론, 그 외 로봇 사용에 대해서는 서비스팀 통해서 지원 받으시면 됩니다. 일단은 저희가 알려드릴 수 있는 부분은 최대한 알려드리도록 하겠습니다.

현재 0.7cm의 오차가 발생한다는 것이 어떤 부분을 말씀하시는 건지요? 현재 TCP는 어떤 값으로 설정하신건가요?
로봇은 설정한 TCP 끝단을 기준으로 움직입니다. TCP를 입력하지 않으시면, Flange 단면의 Center를 중심으로 움직이구요.

문의하신 TCP Calibration은  Workcell > + > Tools 에서 해당되는 Tool을 생성하시고, 특별한 경우가 아니면 Custom Tool을 선택하시면 됩니다. 거기에 수동으로 TCP 길이/각도를 입력하거나 Auto-Calculation 을 통해서 4점을 티칭해서 Calibration할 수 있습니다.

VIduhyeon, 2021-01-20:
@elton-choi 
안녕하세요. 빠른 답변 감사드립니다.

> 현재 0.7cm의 오차가 발생한다는 것이 어떤 부분을 말씀하시는 건지요? 현재 TCP는 어떤 값으로 설정하신건가요?
로봇은 설정한 TCP 끝단을 기준으로 움직입니다. TCP를 입력하지 않으시면, Flange 단면의 Center를 중심으로 움직이구요.

먼저 따로 설정한 TCP값은 없습니다. 툴에 의한 오차를 배제하기 위해 현재 제거한 상태입니다. 따라서 말씀하신대로 Flange단면의 center를 중심으로 움직이게 될텐데요. 같은 목표 지점(x:-54, y:236, z:1316, Rx :0, Ry : 0, Rz : 0)에 대한 서로 다른 Joint Configuration으로 로봇을 이동시켰을때, Flange단면의 center 위치가 y방향으로 0.7cm의 오차가 있다는 것입니다. 해당 오차의 측정 방식은 아래의 그림을 참고해주세요.

밑의 그림을 참고해 주세요.
<로봇팔 실험 환경>
[image]

<오차 측정 방법> - 해당 그림은 로봇팔을 Z축을 제외한 X,Y평면으로 보았을때를 나타내었습니다. 
그에 대한 위치 오차 결과는 제가 올린 위 질문의 < TASK상 같은 좌표에 대한 Joint Configuration 1> < TASK상 같은 좌표에 대한 Joint Configuration 2>를 참고해 주세요.
[image]

> 일단 로봇의 반복정밀도는 스펙에 있으나 정확도에 대해서는 스펙으로 보장하지는 않아서 공식 지원을 받기는 어려우실 것 같습니다. 

말씀하신것이 반복정밀도의 경우, 동일한 Joint Configuration에 대해 반복적으로 로봇을 움직였을때 Flange의 오차가 +-0.05mm라는 말씀이고, 같은 TASK좌표에 대해 여러 Joint Configuration에 대한 정확도는 보장하지 않는다는 말씀으로 이해 했습니다.
제가 이해한 것이 맞는지 답변해주신다면 감사하겠습니다.

closed 2021-04-08 by doosan-robotics