DoosanRobotics/doosan-robot #67: Problem controling A0912 in virtual mode

sdfzz, 2021-01-21:
Hi,

I'm trying to control a doosan A0912 using Moveit in virtual mode. Under ROS Melodic and Ubuntu 18.04, I was able to install doosan ros package. However, when I tried:

```
roslaunch dsr_control dsr_moveit.launch model:=a0912 mode:=virtual
```
and
```
ROS_NAMESPACE=/dsr01a0912 rosrun moveit_commander moveit_commander_cmdline.py robot_description:=/dsr01a0912/robot_description
>> use arm
>> goal = [1 1 1 1 1 1]
>> go goal
```
The model in rviz moves a tiny bit and stops with the following message:
```
[ INFO] [1611228176.220449032]: [trajectory] [16 : 1.227]  38.198  38.196  38.196  38.195  38.198  38.195
[ INFO] [1611228176.220489883]: [trajectory] [17 : 1.283]  40.586  40.583  40.583  40.582  40.585  40.582
[ INFO] [1611228176.220538575]: [trajectory] [18 : 1.342]  42.973  42.970  42.970  42.969  42.973  42.969
[ INFO] [1611228176.220578496]: [trajectory] [19 : 1.406]  45.361  45.358  45.358  45.356  45.360  45.356
[ INFO] [1611228176.220616157]: [trajectory] [20 : 1.476]  47.748  47.745  47.745  47.743  47.747  47.744
[ INFO] [1611228176.220657286]: [trajectory] [21 : 1.554]  50.135  50.132  50.132  50.131  50.135  50.131
[ INFO] [1611228176.220705062]: [trajectory] [22 : 1.646]  52.523  52.519  52.519  52.518  52.522  52.518
[ INFO] [1611228176.220753403]: [trajectory] [23 : 1.765]  54.910  54.907  54.906  54.905  54.909  54.905
[ INFO] [1611228176.220799677]: [trajectory] [24 : 2.054]  57.298  57.294  57.294  57.292  57.297  57.292
[ INFO] [1611228176.224670735]: On Monitor State
[callback OnMonitoringStateCB] current state: (2) MOVING
[ INFO] [1611228176.420833547]: Controller dsr_joint_trajectory_controller successfully finished
[ INFO] [1611228176.718987444]: Completed trajectory execution with status SUCCEEDED ...
[ INFO] [1611228184.073465332]: On Monitor State
[callback OnMonitoringStateCB] current state: (1) STANDBY
```
However, if I check the current joint angle using 'rostopic echo /dsr01a0912/joint_states', I get:
```
name: [joint1, joint2, joint3, joint4, joint5, joint6]
position: [0.015902256057811887, 0.015902397538244795, 0.01590126049329501, 0.015900298218291756, 0.015900319024237775, 0.015900677926806554]
```
which is noway near my goal of [1 1 1 1 1 1 1].

An interesting thing is, If I try the same command with a m0609 instead, 
```
roslaunch dsr_control dsr_moveit.launch model:=m0609 mode:=virtual
```
and
```
ROS_NAMESPACE=/dsr01m0609 rosrun moveit_commander moveit_commander_cmdline.py robot_description:=/dsr01m0609/robot_description 
>> use arm 
>> goal = [1 1 1 1 1 1]
>> go goal 
```
The robot successfully moves to joint position [1 1 1 1 1 1], as can be confirmed by 'rostopic echo /dsr01m0609/joint_states'.

I believe there might be an error in A0912 package. I have an A0912 and I would like to use it with moveit for my project.
Would you please check if you can reproduce this error?

Sincerely,

Steve

doosan-robotics, 2021-02-02:
We have confirmed your inquiry.
In our test environment, the moveit_commander for A0912 works fine.
[1 1 1 1 1 1] It has been confirmed that it moves normally to the joint coordinate system.

[image]

Please try after reinstalling the package. If you still have problems afterwards, please contact us again.

doosan-robotics, 2021-02-02:
Delete all files in the devel and build directories under catkin_ws and try to build again.

sdfzz, 2021-02-04:
Hi, thank you for the reply

Unfortunately, I've already tried to reinstall the repository, but the problem persists.

However, I've tested the code with the real A0912S, and I can move my robot using Moveit.

So I guess I can use it as it is for now.

Sincerely, 

Steve

doosan-robotics, 2021-03-11:
If the same phenomenon occurs in the future, we will update and correct it.

Thank you.

closed 2021-03-11 by doosan-robotics