DoosanRobotics/doosan-robot #68: Client T/P authentication failed.

kirillmorozov, 2021-02-03:
Hello guys.

I've successfully built `doosan-robot` and now I'm trying to use it with a real a0509s robot.

I'm launching ROS using:
```
roslaunch dsr_control dsr_control.launch mode:=real ns:=NR_ model:=a0509 host:=192.168.31.178
```

and get the following console output:

```
auto-starting new master
process[master]: started with pid [9807]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to fd4fae50-665b-11eb-b6fe-0242ac110002
process[rosout-1]: started with pid [9820]
started core service [/rosout]
process[NR_a0509/controller_spawner-2]: started with pid [9837]
[INFO] [1612383086.821778]: Controller Spawner: Waiting for service controller_manager/load_controller
process[NR_a0509-3]: started with pid [9838]
[ INFO] [1612383086.995807744]: rate is 100

[ INFO] [1612383086.998433128]: name_space is NR_, a0509

[ INFO] [1612383086.998484519]: [dsr_hw_interface] constructed
[ INFO] [1612383087.086180911]: [dsr_hw_interface] init() ==> setup callback fucntion
[ INFO] [1612383087.086213272]: INIT@@@@@@@@@@@@@@@@@@@@@@@@@
[ INFO] [1612383087.086229644]: [dsr_hw_interface] init() ==> arm is standby
[ INFO] [1612383087.090174037]: host 192.168.31.178, port=12345 bCommand: 1, mode: real

No Emul
[WARN] [1612383117.118992]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[NR_a0509/controller_spawner-2] process has finished cleanly
```

at the same time robot's teach pendant displays the following message:

```[1.3004] Client T/P authentication failed.```

I have not found any info regarding this message on the internet nor in manuals.

What am I doing wrong? And how do I fix this?

doosan-robotics, 2021-02-04:
It is considered a communication error.
After restarting the controller, complete mastering and try to connect again.
If connection is not possible afterwards, attach the launch file and log in use.

kirillmorozov, 2021-02-04:
I've restarted the controller, but was not able to connect to my robot.
I use this [launch file](https://github.com/doosan-robotics/doosan-robot/blob/788e00128561bf5275c02597bb5bdf1ff45849f8/dsr_control/launch/dsr_control.launch) and start ROS using `roslaunch dsr_control dsr_control.launch mode:=real ns:=NR_ model:=a0509 host:=192.168.31.178`.
Here is full [launch.log](https://github.com/doosan-robotics/doosan-robot/files/5923649/launch.log).
BTW I'm using code from current master branch.

doosan-robotics, 2021-02-04:
1. Test the connection between the controller and the ROS PC using the ping command.
> ping 192.168.31.178

If you can't connect, set up the network again. The controller and ROS PC must be within the same bandwidth.

2. Execute the launch file of the dsr_launcher package.
>roslaunch dsr_launcher single_robot_rviz.launch mode:=real host:=192.168.31.178 model:=a0509 ns:=NR_

3. If it is connected normally, the launch file will be executed and the robot motor will be powered on.

kirillmorozov, 2021-02-04:
Ping command output:

```
PING 192.168.31.178 (192.168.31.178) 56(84) bytes of data.
64 bytes from 192.168.31.178: icmp_seq=1 ttl=63 time=0.349 ms
64 bytes from 192.168.31.178: icmp_seq=2 ttl=63 time=0.337 ms
64 bytes from 192.168.31.178: icmp_seq=3 ttl=63 time=0.355 ms
64 bytes from 192.168.31.178: icmp_seq=4 ttl=63 time=0.462 ms
64 bytes from 192.168.31.178: icmp_seq=5 ttl=63 time=0.468 ms
64 bytes from 192.168.31.178: icmp_seq=6 ttl=63 time=0.468 ms
64 bytes from 192.168.31.178: icmp_seq=7 ttl=63 time=0.335 ms
^C
--- 192.168.31.178 ping statistics ---
7 packets transmitted, 7 received, 0% packet loss, time 6131ms
rtt min/avg/max/mdev = 0.335/0.396/0.468/0.062 ms
```
Attempt to launch `single_robot_rviz` still leads to `[1.3004] Client T/P authentication failed.` message on TP.

doosan-robotics, 2021-02-08:
@kirillmorozov 
Does the same phenomenon occur even if you try to connect again after performing Homming on the TP screen?

doosan-robotics, 2021-04-08:
If you use the Doosan ROS package on a virtual machine, an error (authentication failed) occurs.

closed 2021-04-08 by doosan-robotics