DoosanRobotics/doosan-robot #69: Connection to localhost refused

yuchango, 2021-03-10:
Hi.
I just built the repository (branch: melodic-devel) on a fresh Ubuntu 18.04 and when trying to run
`rosalunch drs_launcher single_robot_rviz.launch `
or
`rosalunch drs_control dsr_moveit.launch `

I keep getting the error
`Connection refused: 127.0.0.1:12345
Net Exception: Socket is not connected`

Could you help me debug this?

Thanks a lot in advance.

yuchango, 2021-03-10:
```
[ INFO] [1615368784.850399217]: INIT@@@@@@@@@@@@@@@@@@@@@@@@@
[ INFO] [1615368784.850408746]: [dsr_hw_interface] init() ==> arm is standby
[ INFO] [1615368784.851290127]: host 127.0.0.1, port=12345 bCommand: 1, mode: virtual

Connection refused: 127.0.0.1:12345
Net Exception: Socket is not connected
[ERROR] [1615368784.915178865]: [dsr_control] Error initializing robot
[dsr01m1013-4] process has died [pid 4688, exit code 255, cmd /home/tx-dev/Doosan_ws/devel/lib/dsr_control/dsr_control_node __name:=dsr01m1013 __log:=/home/tx-dev/.ros/log/9923bb5e-8183-11eb-99ee-7085c23d9e8d/dsr01m1013-4.log].
log file: /home/tx-dev/.ros/log/9923bb5e-8183-11eb-99ee-7085c23d9e8d/dsr01m1013-4*.log
/home/tx-dev/Doosan_ws/src/doosan-robot/common/bin/DRCF/run_drcf.sh: line 23:  4686 Segmentation fault      (core dumped) ./DRCF64 $1 $2
================================================================================REQUIRED process [dsr01/drcf-2] has died!
process has died [pid 4683, exit code 139, cmd /home/tx-dev/Doosan_ws/src/doosan-robot/common/bin/DRCF/run_drcf.sh 12345 m1013 __name:=drcf __log:=/home/tx-dev/.ros/log/9923bb5e-8183-11eb-99ee-7085c23d9e8d/dsr01-drcf-2.log].
log file: /home/tx-dev/.ros/log/9923bb5e-8183-11eb-99ee-7085c23d9e8d/dsr01-drcf-2*.log
Initiating shutdown!
================================================================================
```

doosan-robotics, 2021-03-11:
On melodic, the emulator is connected normally.
Please check it as it may be a problem with the user development environment.
Or, please check the following.

1. Does the common/bin/DRCF/DRCF64 binary file run normally?
2. Does the common/bin/DRCF/run_drcf.sh script run normally?

Thank you.

yuchango, 2021-03-12:
Hi. Thank you for your reply.

DRCF64 is not running properly. The following is the output using gdb:
`Starting program: /home/tx-dev/Doosan_ws/src/doosan-robot/common/bin/DRCF/DRCF64 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
_______________________________________________________________
        Emulator of Doosan Robot Controller [ver M2.52]               
                  [service port = 12345]                          
                  [robot model = m1013(config/config.xml)]                       
_______________________________________________________________
[New Thread 0x7ffff6e83700 (LWP 4529)]

Thread 2 "DRCF64" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff6e83700 (LWP 4529)]
0x00000000004de7a0 in DRAFramework::CTaskInterface::IsTaskRunning() ()
(gdb) bt
#0  0x00000000004de7a0 in DRAFramework::CTaskInterface::IsTaskRunning() ()
#1  0x000000000049f90a in DRAFramework::CTaskScheduler::IsRunning() ()
#2  0x00000000004df522 in TaskIdle(void*) ()
#3  0x00000000006c01db in Poco::ThreadImpl::callableEntry(void*) ()
#4  0x00007ffff7bbb6db in start_thread (arg=0x7ffff6e83700) at pthread_create.c:463
#5  0x00007ffff6fa571f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
`

Do you have any idea what could be the problem?
Thanks in advance.

doosan-robotics, 2021-03-12:
Please tell us more about your development environment.

1. Linux version
2. Kernel version
3. Architecture
4. Operating system bit

In our development environment, the emulator works normally.

yuchango, 2021-03-12:
Right, sorry about that.
1. Ubuntu 18.04.5 LTS
2. Kernel 4.15.0-136-generic
3. x86_64
4. 64-bit

It's running on an Intel Core i7-7700 @ 3.60GHz and 15.6GiB RAM
Please let me know if youneed any more information.

pak-da, 2021-03-13:
Try to launch as root. So type `sudo -i` go in the catkin_ws `source devel/setup.bash` and lauch with roslauch. This should work.

EtienneSchmitz, 2021-03-13:
I have the same problem before, but I found the solution to launch the emulator in sudo mode (in the script)

pak-da, 2021-03-13:
I finally solved it with the following steps:
1) Modify the script ` doosan-robot/common/bin/DRCF/run_drcf.sh ` in order to execute the binary DRCFXX as root:
-  `sudo ./DRCF64` instead of `./DRCF64`
2) In order to run the previous script as root but wo password request /etc/sudoers.tmp needs to be edit with visudo: 
- `$ sudo visudo`
- Add the following command:
  `_user_ ALL=(ALL) NOPASSWD: /home/_user_/catkin_ws/src/doosan-robot/common/bin/DRCF/DRCF64`
- save and exit

Now the launch files in dsr_launcher should work properly. 
I had this problem suddenly. Some days ago I've launched the launch files without these modifications.

doosan-robotics, 2021-03-15:
The emulator works only under **sudo permission**.
Thank you for answer.

yuchango, 2021-03-15:
Hi everyone.

Using sudo indeed solved the problem.

Thanks a lot for your support!

closed 2021-03-15 by yuchango