DoosanRobotics/doosan-robot #70: Cannot modify end-effector rotation with jog_multi rosservice

yuchango, 2021-04-14:
Hi.
I am trying to test the jog service with the robot simulation, and am having some problems:

1. I cannot modify the end-effector orientation using motion/jog_multi service.
I try calling 
`rosservice call /<model>/motion jog_multi -- [0,0,0,1,0,0] 0 50`
but nothing happens.
Could you guide me through what I am doing incorrectly?

2. Is there a way to know if the jog or jog_multi services will or will not work?
I has happened to me several times that depending on the configuration of the robot arm, sometimes calling the service does absolutely nothing, even though the motion is possible.
One clear example is when trying to call for instance:
`rosservice call /<model>/motion jog_multi -- [0,0,-1,0,0,0] 0 50`
i.e. a downward vertical motion, with the robot being in zero position (with the arm completely extended upward), even though the robot should be able to move down, it does nothing and stays the same.
Is this related to the robot being close to singularities or difficult configurations?
Is there a way to solve this problem?

I am on the melodic-devel branch on an Ubuntu18.04 machine.
I would appreciate any guidance you could give me.
Thanks in advance.

doosan-robotics, 2021-04-14:
1. Currently, the jog_multi function is provided using rostopic, not rosservice. Publish using rostopic, or use the single-axis jog feature.

2. First, check if jog is working properly. The single-axis jog allows for singularity avoidance. However, since the multi-axis jog (jog_multi) function is not an officially supported function, vibration may occur near the singular point.

yuchango, 2021-04-14:
Thank you very much for your reply.

I tried using the rostopic, but the behavior is still the same.
1. The single axis jog works well, but when trying to modify the end-effector orientation with jog_multi either through rosservice or rostopic, the robot does not move.
2. I tried using jog_multi thorugh a rostopic, but it still does not move at all in the described situation 
(trying to call for instance:
`rosservice call /<model>/motion jog_multi -- [0,0,-1,0,0,0] 0 50`
i.e. a downward vertical motion, with the robot being in zero position (with the arm completely extended upward), even though the robot should be able to move down, it does nothing and stays the same.)
Is there anyway to solve these problems, or am I doing something incorrectly?

Thanks!

doosan-robotics, 2021-04-15:
Sorry, I checked again, and I found the rotation part in jog_multi doesn't work.
I didn't noticed it, since this function is not supported officially.
We are currently developing general speedj/speedl function this year, and you can use it as multi jog function without manual speed limit(250mm/s, 30deg/s).

yuchango, 2021-04-15:
Thanks for checking.

Roger about the speedj/speedl functions. When is it planned to be ready?

doosan-robotics, 2021-04-19:
It will be released on November this year. Thank you.

closed 2021-10-19 by doosan-robotics