DoosanRobotics/doosan-robot #71: unkonw robot model config in the emulator

pak-da, 2021-04-15:
Hi. 
I am trying to run the emulator with the a0509 but it seems that it doesn't find the configuration file.
`sudo ./DRCF64 12345 a0509` release me the output in the following image. 
[image]

Do you have any suggestion? Other model seem work fine.

doosan-robotics, 2021-04-16:
There is no problem with the current behavior and will be reflected in future emulator updates.

pak-da, 2021-04-16:
I thought it was a problem because putting all the joints to zero the end effector position doesn't match the real position/the position on the teach pendant.
The first image is the teach pendant of a a0509 the second one is the state topic with the a0509 during the simulation.
Do you have any idea about this mismatch please?
[image]
[image]

doosan-robotics, 2022-06-15:
As a result of the check, it was confirmed that the A0509 modeling file inserted in the emulator source was abnormal. It will be corrected later.

closed 2022-06-15 by doosan-robotics