DoosanRobotics/doosan-robot #74: Deactivate robot alarms through ROS

yuchango, 2021-05-05:
I wanted to know if there is a way to regain control of the robot after the servos are turned off due to System alarms, such as collisions, joints going outside joint or torque limits, etc.
Specifically, I'd like to be able to turn on the servos and regain control of  the robot through ROS, be it a rosservice, sending a message to a rostopic, etc. Is this possible?

Thank you very much in advance!

doosan-robotics, 2021-05-09:
Currently, the service for Servo On and control right recovery is not implemented in the ROS package.
It is implemented on the API, so please refer to the __dsr_hw_interface.cpp__ file of the dsr_control package.
**Access control API function: Drfl.ManageAccessControl**
**Servo On API Function: Drfl.SetRobotControl**

yuchango, 2021-05-10:
Thank you very much for your reply. I will check those functions.

Cheers!

closed 2021-05-10 by yuchango