DoosanRobotics/doosan-robot #75: Robot and PC time synchronization

yuchango, 2021-05-05:
I am trying to use a robot trajectory generator to use with the real robot, which basically creates a message of type `trajectory_msgs/JointTrajectory`, and  publishes it to the topic `/dsr01h2515/dsr_joint_trajectory_controller/command`
The problem is that as soon as the topic is published, the following error appears:

`[ WARN] [1620190966.815448805]: Dropping all 1 trajectory point(s), as they occur before the current time.
Last point is XXXXs in the past.
`
Also, I wanted to try first in virtual mode, but in that mode, I cannot load joint controllers of type FollowJointTrajectory.
Is this a synchronization problem? Could you help me solve this problem?

Thanks a lot in advance!

doosan-robotics, 2021-05-09:
When using the m-series robot, it operates normally (without delay).
Please check the point below.
1. Is there always a 2 second delay?
2. How many trajectory points do you encounter?
3. Can you attach log files or video files?

Thank you.

yuchango, 2021-05-10:
Thank you very much for your reply.

I am very sorry. I just realized that what was mainly causing the delay was that `m_sub_joint_trajectory` had a buffer of 10 messages. Reducing this to one reduced the delay considerably.

Even though I managed to reduce the delay, I'd like to achieve a faster response.
As I am using the FollowJoinbtTrajectoryAction, I think I am constrained by the parameter `time_from_start`, which cannot be zero. So I have some questions:

- Is there a faster way to move the robot's joints? 
- Would using the positionCallback or jogCallback be faster? 
- If these other functions are faster, do you have an idea how faster would they be?

Thanks a lot in advance!

doosan-robotics, 2021-08-10:
There is currently no way to move faster.

It will also work a bit faster in positionCallback than in jogCallback in terms of robot function (jog has speed limit).

PR #106 (2023-02-24): Use movej instead of movesj for trajectory execution
In this PR we propose a different way of handling the jointTrajectoryAction by using movej instead of movesj.
This removes the 100 point limit, and fixes #75 and #101.
closed 2021-10-19 by doosan-robotics