DoosanRobotics/doosan-robot #78: Robot controller stops working after X number of system alarms

yuchango, 2021-06-02:
Hi.

I have been successfully using your robot using the robot through
`roslaunch dsr_control dsr_moveit.launch model:=h2515 mode:=real host:=<robot_IP>`
and sending joint_position_controller commands.

Then, as I realized the robot would sometimes disable its motors given collisions, going over limits, etc., I added a function to restart all the arm servos after any alarm at runtime, which is only calling the function
` Drfl.SetRobotControl((ROBOT_CONTROL)CONTROL_SERVO_ON);`

However, I've noted that after performing the loop of the robot disabling its servos and enabling them again a couple of times, at some point the controller just stops working. The robot mainly stops with alarms 7170 and 7060 (level 2, group 5).

Do you have any idea why is this happening? Is there any counter that will kill the controller after a given number of alarms? 

Thanks a lot in advance!

doosan-robotics, 2021-08-10:
The 7060 error is an error that occurs when a collision is detected. Please check the robot status.
Sorry for the late reply.

duygiang1162, 2024-04-04:
Hei guy. How to fix this problem?