DoosanRobotics/doosan-robot #83: Error when running m0609

BamBamBap, 2021-08-04:
Hello, An error occurs when executing m0609. Is there any way to solve it?
Below is the error code and rviz picture.
I am using melodic, ubuntu 18.04.

`kim@kim:~$ roslaunch dsr_launcher single_robot_rviz.launch mode:=virtual model:=m0609
... logging to /home/kim/.ros/log/0d93ac3e-f4c9-11eb-9f98-40b076dd2030/roslaunch-kim-7274.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://kim:38303/

SUMMARY
========

PARAMETERS
 * /dsr01/dsr_joint1_position/joint: joint1
 * /dsr01/dsr_joint1_position/pid/d: 200.0
 * /dsr01/dsr_joint1_position/pid/i: 100.0
 * /dsr01/dsr_joint1_position/pid/i_clamp: 500
 * /dsr01/dsr_joint1_position/pid/p: 500.0
 * /dsr01/dsr_joint1_position/type: position_controll...
 * /dsr01/dsr_joint2_position/joint: joint2
 * /dsr01/dsr_joint2_position/pid/d: 250.0
 * /dsr01/dsr_joint2_position/pid/i: 100.0
 * /dsr01/dsr_joint2_position/pid/i_clamp: 500
 * /dsr01/dsr_joint2_position/pid/p: 500.0
 * /dsr01/dsr_joint2_position/type: position_controll...
 * /dsr01/dsr_joint3_position/joint: joint3
 * /dsr01/dsr_joint3_position/pid/d: 40.0
 * /dsr01/dsr_joint3_position/pid/i: 20.0
 * /dsr01/dsr_joint3_position/pid/i_clamp: 500
 * /dsr01/dsr_joint3_position/pid/p: 300.0
 * /dsr01/dsr_joint3_position/type: position_controll...
 * /dsr01/dsr_joint4_position/joint: joint4
 * /dsr01/dsr_joint4_position/pid/d: 40.0
 * /dsr01/dsr_joint4_position/pid/i: 0.0
 * /dsr01/dsr_joint4_position/pid/i_clamp: 500
 * /dsr01/dsr_joint4_position/pid/p: 100.0
 * /dsr01/dsr_joint4_position/type: position_controll...
 * /dsr01/dsr_joint5_position/joint: joint5
 * /dsr01/dsr_joint5_position/pid/d: 25.0
 * /dsr01/dsr_joint5_position/pid/i: 20.0
 * /dsr01/dsr_joint5_position/pid/i_clamp: 500
 * /dsr01/dsr_joint5_position/pid/p: 200.0
 * /dsr01/dsr_joint5_position/type: position_controll...
 * /dsr01/dsr_joint6_position/joint: joint6
 * /dsr01/dsr_joint6_position/pid/d: 25.0
 * /dsr01/dsr_joint6_position/pid/i: 20.0
 * /dsr01/dsr_joint6_position/pid/i_clamp: 500
 * /dsr01/dsr_joint6_position/pid/p: 300.0
 * /dsr01/dsr_joint6_position/type: position_controll...
 * /dsr01/dsr_joint_position_controller/joints: ['joint1', 'joint...
 * /dsr01/dsr_joint_position_controller/publish_rate: 20
 * /dsr01/dsr_joint_position_controller/type: position_controll...
 * /dsr01/dsr_joint_publisher/publish_rate: 150
 * /dsr01/dsr_joint_publisher/type: joint_state_contr...
 * /dsr01/dsr_joint_trajectory_controller/joints: ['joint1', 'joint...
 * /dsr01/dsr_joint_trajectory_controller/publish_rate: 20
 * /dsr01/dsr_joint_trajectory_controller/type: position_controll...
 * /dsr01/dsr_velocity_controller/angular/z/has_acceleration_limits: True
 * /dsr01/dsr_velocity_controller/angular/z/has_velocity_limits: True
 * /dsr01/dsr_velocity_controller/angular/z/max_acceleration: 6.0
 * /dsr01/dsr_velocity_controller/angular/z/max_velocity: 2.0
 * /dsr01/dsr_velocity_controller/base_frame_id: base_link
 * /dsr01/dsr_velocity_controller/cmd_vel_timeout: 0.25
 * /dsr01/dsr_velocity_controller/enable_odom_tf: False
 * /dsr01/dsr_velocity_controller/estimate_velocity_from_position: False
 * /dsr01/dsr_velocity_controller/left_wheel: ['front_left_whee...
[... 174 lines cut ...]
  File "/home/kim/.local/lib/python2.7/site-packages/numpy/ma/__init__.py", line 44, in <module>
    from . import core
  File "/home/kim/.local/lib/python2.7/site-packages/numpy/ma/core.py", line 6556, in <module>
    class _frommethod(object):
KeyboardInterrupt
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 38, in <module>
    import rospy, sys
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/__init__.py", line 47, in <module>
    from std_msgs.msg import Header
  File "/opt/ros/melodic/lib/python2.7/dist-packages/std_msgs/msg/__init__.py", line 3, in <module>
    from ._ByteMultiArray import *
  File "/opt/ros/melodic/lib/python2.7/dist-packages/std_msgs/msg/_ByteMultiArray.py", line 2, in <module>
    """autogenerated by genpy from std_msgs/ByteMultiArray.msg. Do not edit."""
KeyboardInterrupt
/home/kim/catkin_ws/src/doosan-robot/dsr_launcher/scripts/dmodel.py:28: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  self.data = yaml.load(base_file)
[dmodel-9] escalating to SIGTERM
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`
[image]

EtienneSchmitz, 2021-08-04:
Hi,

I have the same problem than you before. I correct it with this method :
1. Modify /home/kim/catkin_ws/src/doosan-robot/common/bin/DRCF/run_drcf.sh and put sudo before this to launch DRCF64 in sudo mode.
2. With sudo visudo, add this line at the end of the file :
```bash
username ALL= NOPASSWD:/home/username/ros_workspace/src/doosan-robot/common/bin/DRCF/DRCF64
```

BamBamBap, 2021-08-04:
Thank you so much! I solved it!

closed 2021-08-10 by doosan-robotics