DoosanRobotics/doosan-robot #85: Connection to real robot from within a Docker container

mpavezb, 2021-08-24:
Hi,

I am not able to control the robot using ROS from within a Docker container. I've been following the instructions in the manual and from this repo.

For ROS noetic I am getting:
```bash
$ roslaunch dsr_launcher single_robot_gazebo.launch model:=m0609 color:=white mode:=real host:=192.168.137.100
...
[ INFO] [1629839326.375662912]: [dsr_hw_interface] init() ==> constructed
[ INFO] [1629839326.455662912]: [dsr_hw_interface] init() ==> setup callback fucntion
[ INFO] [1629839326.455688922]: INIT@@@@@@@@@@@@@@@@@@@@@@@@@
[ INFO] [1629839326.455701092]: [dsr_hw_interface] init() ==> arm is standby
[ INFO] [1629839326.459429688]: host 192.168.137.100, port=12345 bCommand: 1, mode: real
...
No errors were displayed.
```
Results for ROS foxy are similar.

On the DART Platform the following log message is displayed: `[1.3004] Client T/P authentication failed.`. The same as for this previous issue: https://github.com/doosan-robotics/doosan-robot/issues/68

Given the previous info, I am assuming this issue is similar to https://github.com/doosan-robotics/doosan-robot/issues/68

Is it possible to use this package from within Docker?

mpavezb, 2021-08-25:
Linking similar issue for ROS2: https://github.com/doosan-robotics/doosan-robot2/issues/14

doosan-robotics, 2021-09-29:
Sorry for the late reply.
What version of the robot are you using?
ex. 2.8.

mpavezb, 2021-10-14:
The robot version is 2.7.

closed 2022-08-31 by doosan-robotics

doosan-robotics, 2023-03-02:
@mpavezb 
Please refer to https://github.com/doosan-robotics/doosan-robot2/issues/14