DoosanRobotics/doosan-robot #86: Coordinates of the flange not matching ROS package - A0912

PierreExeter, 2021-10-04:
Hello,

We have a Doosan A0912 and we want to get the coordinates of the flange using ROS.

When the robot is in the home position (vertical), the flange should be at (0, 39, 1395), however ROS returns (0, 34.5 1452.5).
Do you know what causes this mismatch?

I'm using ROS Noetic and here's the command I'm using:
```
roslaunch dsr_launcher single_robot_gazebo.launch mode:=virtual model:=a0912
```

Here is how I retrieve the flange position:
```
import rospy

current_robot_state = rospy.wait_for_message('/dsr01a0912/state', RobotState)
print(current_robot_state.current_tool_posx)
```

Thanks

PierreExeter, 2021-10-04:
I realised that the values I get are from the M1013 robot, not the A0912. However, the robot shown in Gazebo is the A0912.
Could it be that the robot model is being overwritten somewhere in the ROS package? (maybe in the RobotState message)

nicola-sysdesign, 2021-12-21:
I have the same issue with the H2017 model. I checked and they seem to be missing config files for H-Series models too. Can you add them so that emulator works correctly for the H-Series models too?

doosan-robotics, 2022-01-12:
ROS emulator development is delayed due to API updates. We plan to release a new DRCF emulator in February, which will include both A / H series.

closed 2024-01-30 by doosan-robotics