DoosanRobotics/doosan-robot #92: Robot returns `FollowJointTrajectoryActionResult` before move is executed

robertdagenais, 2021-11-11:
Hello! I am trying to move the h2017 robot in "real mode" using the Doosan ROS drivers. When I send move commands using MoveIt
(for example:
```
move_group.go([0, 0, 1.57, 0, 1.57, 0], wait=True)
move_group.go([0, 0, 0, 0, 0, 0], wait=True)
```
), it seems the robot controller returns successful and starts planning the next move BEFORE the previous move has even finished.

In my understanding, MoveIt sends a request to the `FollowJointTrajectoryAction` service [here](https://github.com/ros-planning/moveit/blob/4db626d9c3b5d6296b012188a4a0bfe4bddf6bce/moveit_plugins/moveit_simple_controller_manager/src/follow_joint_trajectory_controller_handle.cpp#L65) and the ROS driver executes a callback when the `FollowJointTrajectoryActionGoal` is received [here](https://github.com/doosan-robotics/doosan-robot/blob/faebadaf12dee13578621720931552f2d1298af8/dsr_control/src/dsr_hw_interface.cpp#L1084). According to the logs (displayed below), it appears that the [controllerDoneCallback](https://github.com/ros-planning/moveit/blob/4db626d9c3b5d6296b012188a4a0bfe4bddf6bce/moveit_plugins/moveit_simple_controller_manager/src/follow_joint_trajectory_controller_handle.cpp#L197) executes (returning success) before the move is actually finished - observe `Controller dsr_joint_trajectory_controller successfully finished` before the robot controller announces it is done moving and is in `current state: (1) STANDBY`.

[image]

Is this the expected behaviour -> for the robot controller to return `FollowJointTrajectoryActionResult` successfully once the goal is received BUT before it has executed on the real robot?

doosan-robotics, 2022-01-12:
The robot will perform spline joint motion using a waypoint.

closed 2022-01-12 by doosan-robotics

gavanderhoorn, 2022-01-12:
@doosan-robotics: could you clarify how this answers @robertdagenais observation?

The "contract" between a `FollowJointTrajectoryAction` server and client is that the client submits a goal, the server executes this (to the best of its ability) and the server notifies the client if/when it has completed execution.

For this to work correctly (for instance to be able to execute trajectories back-to-back), it's imperative the server notifies the client about completion at the correct time.

From what @robertdagenais describes, the current implementation returns `SUCCESS` after having received the goal, not after having completely executed the trajectory.