DoosanRobotics/doosan-robot #113: Error launching roslaunch dsr_launcher dsr_moveit.launch model:=m1509 gripper:=robotiq_2f

juju0111, 2022-06-26:
안녕하세요. $ roslaunch dsr_launcher dsr_moveit.launch model:=m1509 gripper:=robotiq_2f 를 실행하면 다음과 같은 에러가 납니다. 

[INFO] [1656219292.552720]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1656219292.555400]: Loading controller: dsr_joint_trajectory_controller
[ERROR] [1656219292.566770892]: Could not find joint 'robotiq_85_left_knuckle_joint' in URDF model.
[ERROR] [1656219292.566829916]: Failed to initialize the controller
[ERROR] [1656219292.566847470]: Initializing controller 'dsr_joint_trajectory_controller' failed
[ERROR] [1656219293.568674]: Failed to load dsr_joint_trajectory_controller
[INFO] [1656219293.571613]: Loading controller: dsr_joint_publisher
[INFO] [1656219293.583370]: Controller Spawner: Loaded controllers: dsr_joint_publisher
[INFO] [1656219293.593162]: Started controllers: dsr_joint_publisher
[ERROR] [1656219293.873311566]: Joint 'robotiq_85_left_knuckle_joint' not found in model 'm1509'
[ WARN] [1656219293.873443070]: Joint state with name: "robotiq_85_left_knuckle_joint" was received but not found in URDF
[ERROR] [1656219293.883374498]: Joint 'robotiq_85_left_knuckle_joint' not found in model 'm1509'
[ERROR] [1656219293.893341405]: Joint 'robotiq_85_left_knuckle_joint' not found in model 'm1509'
[ERROR] [1656219293.902958086]: Joint 'robotiq_85_left_knuckle_joint' not found in model 'm1509'
[ERROR] [1656219293.913316207]: Joint 'robotiq_85_left_knuckle_joint' not found in model 'm1509'
[ERROR] [1656219293.923453011]: Joint 'robotiq_85_left_knuckle_joint' not found

해결할 수 있을 방법이 있을까요?

doosan-robotics, 2022-08-24:
Currently, the gripper argument does not apply to the dsr_moveit.launch file (you can input it, but it does not apply to the actual description).

Nelson-PT, 2022-08-29:
@doosan-robotics  Is there any prediction for when this functionality will be available? (not only of the m1509 but for other models as well)

doosan-robotics, 2022-11-29:
It will be updated before the second half of next year.