DoosanRobotics/doosan-robot #115: [ERROR] [1657261448.071353066]: [dsr_control] Error initializing robot

JaeYoungLee79, 2022-07-08:
~$ roslaunch dsr_launcher single_robot_gazebo.launch mode:=virtual
... logging to /home/rut21/.ros/log/91d1aab0-fe86-11ec-9db5-988389262a78/roslaunch-rut21-2893.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/rut21/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://rut21:43431/

SUMMARY
========

PARAMETERS
 * /dsr01/dsr_joint1_position/joint: joint1
 * /dsr01/dsr_joint1_position/pid/d: 200.0
 * /dsr01/dsr_joint1_position/pid/i: 100.0
 * /dsr01/dsr_joint1_position/pid/i_clamp: 500
 * /dsr01/dsr_joint1_position/pid/p: 500.0
 * /dsr01/dsr_joint1_position/type: position_controll...
 * /dsr01/dsr_joint2_position/joint: joint2
 * /dsr01/dsr_joint2_position/pid/d: 250.0
 * /dsr01/dsr_joint2_position/pid/i: 100.0
 * /dsr01/dsr_joint2_position/pid/i_clamp: 500
 * /dsr01/dsr_joint2_position/pid/p: 500.0
 * /dsr01/dsr_joint2_position/type: position_controll...
 * /dsr01/dsr_joint3_position/joint: joint3
 * /dsr01/dsr_joint3_position/pid/d: 40.0
 * /dsr01/dsr_joint3_position/pid/i: 20.0
 * /dsr01/dsr_joint3_position/pid/i_clamp: 500
 * /dsr01/dsr_joint3_position/pid/p: 300.0
 * /dsr01/dsr_joint3_position/type: position_controll...
 * /dsr01/dsr_joint4_position/joint: joint4
 * /dsr01/dsr_joint4_position/pid/d: 40.0
 * /dsr01/dsr_joint4_position/pid/i: 0.0
 * /dsr01/dsr_joint4_position/pid/i_clamp: 500
 * /dsr01/dsr_joint4_position/pid/p: 100.0
 * /dsr01/dsr_joint4_position/type: position_controll...
 * /dsr01/dsr_joint5_position/joint: joint5
 * /dsr01/dsr_joint5_position/pid/d: 25.0
 * /dsr01/dsr_joint5_position/pid/i: 20.0
 * /dsr01/dsr_joint5_position/pid/i_clamp: 500
 * /dsr01/dsr_joint5_position/pid/p: 200.0
 * /dsr01/dsr_joint5_position/type: position_controll...
 * /dsr01/dsr_joint6_position/joint: joint6
 * /dsr01/dsr_joint6_position/pid/d: 25.0
 * /dsr01/dsr_joint6_position/pid/i: 20.0
 * /dsr01/dsr_joint6_position/pid/i_clamp: 500
 * /dsr01/dsr_joint6_position/pid/p: 300.0
 * /dsr01/dsr_joint6_position/type: position_controll...
 * /dsr01/dsr_joint_position_controller/joints: ['joint1', 'joint...
 * /dsr01/dsr_joint_position_controller/publish_rate: 20
 * /dsr01/dsr_joint_position_controller/type: position_controll...
 * /dsr01/dsr_joint_publisher/publish_rate: 150
 * /dsr01/dsr_joint_publisher/type: joint_state_contr...
 * /dsr01/dsr_joint_trajectory_controller/joints: ['joint1', 'joint...
 * /dsr01/dsr_joint_trajectory_controller/publish_rate: 20
 * /dsr01/dsr_joint_trajectory_controller/type: position_controll...
 * /dsr01/dsr_velocity_controller/angular/z/has_acceleration_limits: True
 * /dsr01/dsr_velocity_controller/angular/z/has_velocity_limits: True
 * /dsr01/dsr_velocity_controller/angular/z/max_acceleration: 6.0
 * /dsr01/dsr_velocity_controller/angular/z/max_velocity: 2.0
 * /dsr01/dsr_velocity_controller/base_frame_id: base_link
 * /dsr01/dsr_velocity_controller/cmd_vel_timeout: 0.25
 * /dsr01/dsr_velocity_controller/enable_odom_tf: False
 * /dsr01/dsr_velocity_controller/estimate_velocity_from_position: False
 * /dsr01/dsr_velocity_controller/left_wheel: ['front_left_whee...
 * /dsr01/dsr_velocity_controller/linear/x/has_acceleration_limits: True
 * /dsr01/dsr_velocity_controller/linear/x/has_velocity_limits: True
 * /dsr01/dsr_velocity_controller/linear/x/max_acceleration: 3.0
[... 163 lines cut ...]
[dsr01m1013/controller_spawner-6] killing on exit
[dsr01/controller_spawner-4] killing on exit
[gazebo_gui-3] killing on exit
[gazebo-2] killing on exit
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 212, in <module>
    if __name__ == '__main__': main()
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 105, in main
    rospy.init_node('spawner', anonymous=True)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/client.py", line 323, in init_node
    raise rospy.exceptions.ROSInitException("init_node interrupted before it could complete")
rospy.exceptions.ROSInitException: init_node interrupted before it could complete
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 212, in <module>
    if __name__ == '__main__': main()
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 105, in main
    rospy.init_node('spawner', anonymous=True)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/client.py", line 323, in init_node
    raise rospy.exceptions.ROSInitException("init_node interrupted before it could complete")
rospy.exceptions.ROSInitException: init_node interrupted before it could complete
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

JaeYoungLee79, 2022-07-08:
Dear Sir
I will call.

doosan-robotics, 2022-08-24:
The emulator will be updated. Also, the emulator must be run as a root account.

closed 2022-08-24 by doosan-robotics