DoosanRobotics/doosan-robot #116: System exception: cannot lock mutex

JaeYoungLee79, 2022-07-11:
rut21@rut21:~/catkin_workspace/src/doosan-robot/dsr_control/launch$ roslaunch dsr_control.launch 
... logging to /home/rut21/.ros/log/1e27efc0-00e3-11ed-9f0d-988389262a78/roslaunch-rut21-3598.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/rut21/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://rut21:44793/

SUMMARY
========

PARAMETERS
 * /dsr01m1013/command: True
 * /dsr01m1013/dsr_joint1_position/joint: joint1
 * /dsr01m1013/dsr_joint1_position/pid/d: 200.0
 * /dsr01m1013/dsr_joint1_position/pid/i: 100.0
 * /dsr01m1013/dsr_joint1_position/pid/i_clamp: 500
 * /dsr01m1013/dsr_joint1_position/pid/p: 500.0
 * /dsr01m1013/dsr_joint1_position/type: position_controll...
 * /dsr01m1013/dsr_joint2_position/joint: joint2
 * /dsr01m1013/dsr_joint2_position/pid/d: 250.0
 * /dsr01m1013/dsr_joint2_position/pid/i: 100.0
 * /dsr01m1013/dsr_joint2_position/pid/i_clamp: 500
 * /dsr01m1013/dsr_joint2_position/pid/p: 500.0
 * /dsr01m1013/dsr_joint2_position/type: position_controll...
 * /dsr01m1013/dsr_joint3_position/joint: joint3
 * /dsr01m1013/dsr_joint3_position/pid/d: 40.0
 * /dsr01m1013/dsr_joint3_position/pid/i: 20.0
 * /dsr01m1013/dsr_joint3_position/pid/i_clamp: 500
 * /dsr01m1013/dsr_joint3_position/pid/p: 300.0
 * /dsr01m1013/dsr_joint3_position/type: position_controll...
 * /dsr01m1013/dsr_joint4_position/joint: joint4
 * /dsr01m1013/dsr_joint4_position/pid/d: 40.0
 * /dsr01m1013/dsr_joint4_position/pid/i: 0.0
 * /dsr01m1013/dsr_joint4_position/pid/i_clamp: 500
 * /dsr01m1013/dsr_joint4_position/pid/p: 100.0
 * /dsr01m1013/dsr_joint4_position/type: position_controll...
 * /dsr01m1013/dsr_joint5_position/joint: joint5
 * /dsr01m1013/dsr_joint5_position/pid/d: 25.0
 * /dsr01m1013/dsr_joint5_position/pid/i: 20.0
 * /dsr01m1013/dsr_joint5_position/pid/i_clamp: 500
 * /dsr01m1013/dsr_joint5_position/pid/p: 200.0
 * /dsr01m1013/dsr_joint5_position/type: position_controll...
 * /dsr01m1013/dsr_joint6_position/joint: joint6
 * /dsr01m1013/dsr_joint6_position/pid/d: 25.0
 * /dsr01m1013/dsr_joint6_position/pid/i: 20.0
 * /dsr01m1013/dsr_joint6_position/pid/i_clamp: 500
 * /dsr01m1013/dsr_joint6_position/pid/p: 300.0
 * /dsr01m1013/dsr_joint6_position/type: position_controll...
 * /dsr01m1013/dsr_joint_position_controller/joints: ['joint1', 'joint...
 * /dsr01m1013/dsr_joint_position_controller/publish_rate: 20
 * /dsr01m1013/dsr_joint_position_controller/type: position_controll...
 * /dsr01m1013/dsr_joint_publisher/publish_rate: 150
 * /dsr01m1013/dsr_joint_publisher/type: joint_state_contr...
 * /dsr01m1013/dsr_joint_trajectory_controller/joints: ['joint1', 'joint...
 * /dsr01m1013/dsr_joint_trajectory_controller/publish_rate: 20
 * /dsr01m1013/dsr_joint_trajectory_controller/type: position_controll...
 * /dsr01m1013/dsr_velocity_controller/angular/z/has_acceleration_limits: True
 * /dsr01m1013/dsr_velocity_controller/angular/z/has_velocity_limits: True
 * /dsr01m1013/dsr_velocity_controller/angular/z/max_acceleration: 6.0
 * /dsr01m1013/dsr_velocity_controller/angular/z/max_velocity: 2.0
 * /dsr01m1013/dsr_velocity_controller/base_frame_id: base_link
 * /dsr01m1013/dsr_velocity_controller/cmd_vel_timeout: 0.25
 * /dsr01m1013/dsr_velocity_controller/enable_odom_tf: False
 * /dsr01m1013/dsr_velocity_controller/estimate_velocity_from_position: False
 * /dsr01m1013/dsr_velocity_controller/left_wheel: ['front_left_whee...
 * /dsr01m1013/dsr_velocity_controller/linear/x/has_acceleration_limits: True
 * /dsr01m1013/dsr_velocity_controller/linear/x/has_velocity_limits: True
 * /dsr01m1013/dsr_velocity_controller/linear/x/max_acceleration: 3.0
[... 83 lines cut ...]
process has died [pid 3632, exit code 139, cmd /home/rut21/catkin_workspace/src/doosan-robot/common/bin/DRCF/run_drcf.sh 12345 m1013 __name:=drcf __log:=/home/rut21/.ros/log/1e27efc0-00e3-11ed-9f0d-988389262a78/dsr01-drcf-2.log].
log file: /home/rut21/.ros/log/1e27efc0-00e3-11ed-9f0d-988389262a78/dsr01-drcf-2*.log
Initiating shutdown!
================================================================================
[dsr01m1013-4] killing on exit
[dsr01m1013/controller_spawner-3] killing on exit
[ INFO] [1657521099.442670240]: [dsr_control] shutdown time! (sig = 2)
[ INFO] [1657521099.442766625]: [dsr_control] shutdown time! (sig = 2)
[ INFO] [1657521099.442800329]: [dsr_control] shutdown time! (sig = 2)
[dsr01/drcf-2] killing on exit
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 212, in <module>
    if __name__ == '__main__': main()
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 105, in main
    rospy.init_node('spawner', anonymous=True)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/client.py", line 323, in init_node
    raise rospy.exceptions.ROSInitException("init_node interrupted before it could complete")
rospy.exceptions.ROSInitException: init_node interrupted before it could complete
System exception: cannot lock mutex
[dsr01m1013-4] escalating to SIGTERM
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

JaeYoungLee79, 2022-07-11:
emulator running is ok!
----------------------------------
rut21@rut21:~/Desktop/DRCF$ sudo ./DRCF64
[sudo] password for rut21: 
_______________________________________________________________
        Emulator of Doosan Robot Controller [ver M2.52]               
                  [service port = 12345]                          
                  [robot model = m1013(config/config.xml)]                       
_______________________________________________________________
START ROS service...
START ROS service...

doosan-robotics, 2022-08-24:
Problems can arise if you call the callback function too quickly.

closed 2022-11-29 by doosan-robotics