DoosanRobotics/doosan-robot #120: invers kinematic

Patrick-Sc, 2022-09-13:
I have been trying to calculate the inverse kinematics for the Doosann robot M0609 for a few days. 
J4, J5 and J6 seem to be the same in my calculation and the robot angles.

With the angles J1, J2 and J3 I still have problems to get the right ones.
Probably it is because of the geometrical relation, which I have set wrongly.
Attached the Python code for the calculation of the angles J1-J3.

```python
robot_d1 = 0.135
robot_d2 = 0.00625
robot_d3 = 0.0
robot_d4 = 0.368
robot_d5 = 0.0
robot_d6 = 0.121
robot_d7 = 0.0

robot_a0 = 0.0
robot_a1 = 0.0
robot_a2 = 0.411
robot_a3 = 0.0
robot_a4 = 0.0
robot_a5 = 0.0
robot_a6 = 0.0

robot_alpha0 = 0.0
robot_alpha1 = -pi/2
robot_alpha2 = 0.0
robot_alpha3 = pi/2
robot_alpha4 = -pi/2
robot_alpha5 = pi/2
robot_alpha6 = 0.0

robot_theta1 = 0.0 
robot_theta2 = -pi/2 
robot_theta3 = pi/2 
robot_theta4 = 0.0
robot_theta5 = 0.0
robot_theta6 = 0.0
robot_theta7 = 0.0

def eulerAnglesToRotationMatrix(angle) :     
  R_x = np.array([[1,               0,              	  0               ],
                  [0,               cos(angle[0]),      -sin(angle[0])  ],
                  [0,               sin(angle[0]),      cos(angle[0])   ]])

  R_z1 = np.array([[cos(angle[0]),   -sin(angle[0]),    0               ],
                    [sin(angle[0]),   cos(angle[0]),     0               ],
                    [0,               0,                 1               ]])    

  R_y = np.array([[cos(angle[1]),    0,                 sin(angle[1])  ],
                  [0,                1,                 0               ],
                  [-sin(angle[1]),    0,                cos(angle[1])   ]])
                
  R_z2 = np.array([[cos(angle[2]),   -sin(angle[2]),    0               ],
                    [sin(angle[2]),   cos(angle[2]),     0               ],
                    [0,               0,                 1               ]])                     
                    
  R = np.dot(R_z1, np.dot(R_y, R_z2))
  #R = R_z1 * R_y * R_z2
  # R = np.dot(R_z, np.dot( R_y, R_x ))

  return R

R0g_eval = eulerAnglesToRotationMatrix(np.array([alpha, beta, gamma]))

px = x - (robot_d6 + robot_d7) * R0g_eval[0, 2]
py = y - (robot_d6 + robot_d7) * R0g_eval[1, 2]
pz = z - (robot_d6 + robot_d7) * R0g_eval[2, 2]

# Variante 1
################# J1
# Left arm configuration
# r = math.sqrt(math.pow(px, 2) + math.pow(py, 2) - math.pow(robot_d2, 2))
r = math.sqrt(math.pow(px, 2) + math.pow(py, 2))
s = pz - robot_d1

tmp1 = math.atan2(px, py)
tmp2 = math.atan2(math.sqrt(math.pow(r, 2) - math.pow(robot_d2, 2)), robot_d2)
j1 = tmp1 - tmp2
print("j1_1 OK~: ", j1) # Correct angle but wrong sign (the wrong sign is maybe due to the fact that there are two solutions, but the secound solution is wrong)

# Right arm configuration
tmp1 = math.atan2(px, py)
tmp2 = math.atan2(math.sqrt(math.pow(r, 2) - math.pow(robot_d2, 2)), robot_d2)
tmp3 = tmp2 + math.pi
j1 = tmp3 + tmp2
print("j1_2 OK~: ", j1) # Wrong angle

################# J3
# Ellbow configuration
r = math.sqrt(math.pow(px, 2) + math.pow(py, 2) - math.pow(robot_d2, 2))
s = pz - robot_d1
D = (math.pow(r, 2) + math.pow(s, 2) - math.pow(robot_a2, 2) - math.pow(robot_d4, 2)) / (2 * robot_a2 * robot_d4)

j3 = math.atan2(-math.sqrt(1 - math.pow(D, 2)), D)
print("j3_1 OK: ", j3)  # Correct angle but wrong sign (the wrong sign is due to the fact that there are two solutions)

j3 = math.atan2(+math.sqrt(1 - math.pow(D, 2)), D)
print("j3_2 OK: ", j3)  # Correct angle 

################# J2
r = math.sqrt(math.pow(px, 2) + math.pow(py, 2))
r = math.sqrt(math.pow(px, 2) + math.pow(py, 2) - math.pow(robot_d2, 2))
s = pz
#s = pz - robot_d1

j2 = math.atan2(r, s) - math.atan2(robot_a2 + robot_d4 * cos(j3), robot_d4 * sin(j3))
print("j2: ", j2)

C = (math.pow(r, 2) + math.pow(s, 2) + math.pow(robot_a2, 2) - math.pow(robot_d4, 2)) / (2 * math.sqrt(math.pow(r, 2) + math.pow(s, 2)) * robot_a2)
tmp1 = math.atan2(s, r)
tmp2 = math.atan2(math.sqrt(1 - math.pow(C, 2)), C)
j2 = 0.0
# if s > 0:
j2 = tmp2 - tmp1
print("j2: ", j2) # Is wrong
# if s < 0:
j2 = tmp2 + tmp1
print("j2: ", j2) # Is wrong
```

sugangandhi, 2022-12-21:
Is there no solution yet?

doosan-robotics, 2023-03-06:
@sugangandhi 
Please refer to https://github.com/doosan-robotics/API-DRFL/issues/9

sugangandhi, 2023-03-06:
If i want to calculate the inverse kinematics with our own robot pedestal( inclusive of 2 axis motion ) and total of 8 axis then what is the solution...?
I am trying it in the moveit with new URDF, is that right...? @doosan-robotics

benthebear93, 2023-03-20:
@sugangandhi 

I recommend that you search for open-source solutions for inverse kinematics for robots that involve an additional 2-axis. If you are using Moveit, I suggest creating a custom URDF by combining the Doosan robot URDF with a pedestal, which can act as a prismatic joint. However, if you are planning to implement this on a real robot, the process will be significantly different.

sugangandhi, 2023-03-20:
> @sugangandhi
>
> I recommend that you search for open-source solutions for inverse kinematics for robots that involve an additional 2-axis. If you are using Moveit, I suggest creating a custom URDF by [...]

Actually im using it on real robot. In what way it significantly different.? and is it possible to integerate my servo to the robot controller A0509.?
If it is not possible then what will be the solution.?

benthebear93, 2023-03-20:
@sugangandhi 

dsr_control has specific dsr_hw_interface which only takes 6 dof solution. In that case, you might have to rewrite HW interface to take 8 dof solution. I'm not sure it is possible to do it though since it offers limited information.  

I couldn't think of the possible solution to integrate your servo into A0509 Contoller. Alternative could be seperating servo and robot ROS controller without considering synchronized motion. Using "Servo" motion might be better for following trajectory you sent to robot and servo motor.  

@doosan-robotics  any thought on this?

wenke15183073494, 2023-03-24:
Is this problem solved?

sugangandhi, 2023-07-15:
Still looking for better solution.