DoosanRobotics/doosan-robot #124: Force limit for Joint torque sensor in doosan A0509

foreverbala, 2023-01-11:
I am using compliance control and force control in my application. In this, there is a default value of `10N` force for three translational target forces and `5nm` for three rotational target moments. I want to change this default setting to a minimum value in the joint torque sensor. 
I couldn't find where to change this setting. 
The robots current version is `version v2.9.4`
I don't have the option in my teachpendant (workcell manager). 
Please help me out. @doosan-robotics

doosan-robotics, 2023-03-06:
@foreverbala 
A series models without FTS sensors do not support input of rotational and translational stiffness values.

foreverbala, 2023-03-06:
@doosan-robotics 
Thanks for your reply.

doosan-robotics, 2023-03-08:
@foreverbala 
If you have any problems please reopen this issue.

closed 2023-03-08 by doosan-robotics