DoosanRobotics/doosan-robot #125: Error Initializing Real Robot - dsr_control_node (virtual mode works)

DamAnirban, 2023-01-19:
Hi,

I have a new Doosan A0509s arm. I am trying to run it using ROS. 

The virtual mode works and launches properly when using the following - 
`roslaunch dsr_launcher dsr_moveit.launch model:=a0509`

The problem arises when I am connecting the Real robot. I am running the following - 
`roslaunch dsr_control dsr_moveit.launch model:=a0509 host:=192.168.127.100 mode:=real`

But, I am getting the following errors - 

1. Connection refused
[ERROR] [1674121841.041127378]: [dsr_control] Error initializing robot

2. [dsr01a0509-3] process has died [pid 40413, exit code 255, cmd /home/anirban/catkin_ws/devel/lib/dsr_control/dsr_control_node __name:=dsr01a0509 __log:=/home/anirban/.ros/log/bb3cbe20-97de-11ed-9a96-67be539470cf/dsr01a0509-3.log].
log file: /home/anirban/.ros/log/bb3cbe20-97de-11ed-9a96-67be539470cf/dsr01a0509-3*.log

I am adding the terminal+error details, robot connection details in the files attached. I am also adding the terminal details of the virtual mode which is working ok. I am able to ping to 192.168.127.100 when using the real robot, but ROS is unable to connect to it.

System details - 
Ubuntu 20.04
Ros Noetic

Can you please help me debug this issue?

Thank you.

[Real robot_error.txt](https://github.com/doosan-robotics/doosan-robot/files/10455705/Real.robot_error.txt)
[Virual mode_working.txt](https://github.com/doosan-robotics/doosan-robot/files/10455707/Virual.mode_working.txt)
[connection_image](https://user-images.githubusercontent.com/50565847/213415320-cb4c07ca-e887-497f-864e-37b40bde66f1.png)

DamAnirban, 2023-01-23:
Any Ideas regarding this issue? I have been trying for the past few days to no avail. Any pointers will be really helpful.
Thanks.

AnupamKumarPranav, 2023-01-23:
Hi, I am also facing the same issue regarding setup and connecting the robot arm. Need some help.

sugangandhi, 2023-01-24:
In the IPV4 section you need to add the ROBOT IP. In my case `192.162.1.1` and netmask `255.255.255.0`.  
you can check your ROBOT IP in teach pendent and also check your ethernet cable connection.

DamAnirban, 2023-01-24:
Hi @sugangandhi , thanks for responding. Yes the IP, netmask is already added in the IPV4 section. Its 192.168.127.100 and 255.255.255.0 in my case (same in the network settings in the Robot Teaching Pendant). The error still persists.

sugangandhi, 2023-01-24:
> Hi @sugangandhi , thanks for responding. Yes the IP, netmask is already added in the IPV4 section. Its `192.168.127.100` and `255.255.255.0` in my case (same in the network settings in the Robot Teaching Pendant). I have added the screenshot of the connection. The error still persists.

Sorry for this, Have you successful in `catkin_make`.
May be try with reinstall the complete doosan-ros packages.

DamAnirban, 2023-01-27:
Hi @sugangandhi . There is no problem with `catkin_make`. Reset the IP in the pendant also. Did a complete reinstall of the doosan-ros packages....wasnt able to solve it.

I have added the images of the problem.

sugangandhi, 2023-01-27:
> Hi @sugangandhi . There is no problem with `catkin_make`. Reset the IP in the pendant also. Did a complete reinstall of the doosan-ros packages....wasnt able to solve it.
> 
> pdate - I have added more images of the problem.

[image]
 
first one is my network IP
second is my A0509 IP
Try like this!

DamAnirban, 2023-01-30:
No that did not work. But I solved it finally :-). I cant believe the problem was this -> The robot controller was connected directly to my PC. This was the output of the nmap command. The port is closed for some reason.
[image]

Then checked all ports-
[image]

Clearly, No tcp port is open. So added a network switch in between.

Earlier,
Robot Controller<----------------->PC
Now, 
Robot Controller<---------------network switch--------------->PC.

voila - 
[image]

The port is now open and it works.

closed 2023-01-30 by DamAnirban

Saketh-Vaishya, 2023-06-29:
@DamAnirban Hey is the issue resolved with port switch , can you explain more about the solution and what is the exact problem .. Because i have seen many people are not using port switch . Do you think is this a permanent solution

duygiang1162, 2024-03-31:
How did you fixed this problem? Please help me