DoosanRobotics/doosan-robot #128: Motion over RT API in ROS aborts with unknown error code 7056

tf-maam, 2023-01-30:
During a motion using the RT API with the command `servoj_rt` and also when using ROS `servoj_rt_stream` the motion aborts with the undocumented alarm `level : 2 group : 5 index : 7056` causing a `SAFE_OFF` of the robot controller. Afterwards, `dsr_control_node` seems trying to restart the connection. This may cause a jerky motion of the robot, which may exceed the the velocity or acceleration limits and leading to the alarms `level : 2 group : 2 index : 9014` respectively `level : 2 group : 2 index : 9015`.

```
...
[ INFO] [1675069895.902860051]: On Monitor State
[callback OnMonitoringStateCB] current state: (1) STANDBY
[ INFO] [1675069895.903887666]: On Monitor State
[callback OnMonitoringStateCB] current state: (2) MOVING
[ INFO] [1675069897.735972273]: On Monitor State
[callback OnMonitoringStateCB] current state: (3) SAFE_OFF

[ WARN] [1675069898.249474821]: [callback OnLogAlarm]
[ WARN] [1675069898.249528517]:  level : 2
[ WARN] [1675069898.249552795]:  group : 5
[ WARN] [1675069898.249581135]:  index : 7056
[ WARN] [1675069898.249602482]:  param : 
[ WARN] [1675069898.249632761]:  param : 
[ WARN] [1675069898.249667579]:  param : 

[ INFO] [1675069899.504797874]: On Monitor State
[callback OnMonitoringStateCB] current state: (1) STANDBY
[ INFO] [1675069899.505902978]: On Monitor State
[callback OnMonitoringStateCB] current state: (2) MOVING
[ INFO] [1675069899.525433679]: On Monitor State
[callback OnMonitoringStateCB] current state: (5) SAFE_STOP
...
```

The same motion, provided in the RT API example, works fine on this robot. 
Could you please give details on error code **7056** and a possible solution to overcome this problem.

InigoMoreno, 2023-01-30:
I am also facing the same issues so it would be great to have an answer on this. FYI, error 7056 means  _OPERATION_SAFETY_FUNCTION_SOS_VIOLATION_  as seen in [DRFC.h#L1121](  https://github.com/tecnalia-advancedmanufacturing-robotics/doosan-robot/blob/fa4a6b3f47eb0de2c5f7e44b6b7d7fa2b371d95d/common/include/DRFC.h#L1121). Not sure what that means though...

doosan-robotics, 2023-03-02:
I'm sorry for late reply. It seems that you have already solved it, but I will describe the cause of the problem.

5.7056 means an error about an invalid safety event. Robot safety events are divided into three categories (enter, moving, stop). Distinguishing the three states is straightforward. When in the moving state, the robot LEDs (on the robot flange and base) blink. 

In general, our commands automatically adjust safety events, but real-time control commands are an exception. The user must manually set the safety event for real-time control such as servoj.

The user can use set_safety_mode command for change safety event.
Please following below example code.

Drfl.set_safety_mode(SAFETY_MODE_AUTONOMOUS, SAFETY_MODE_EVENT_MOVE); 

/// servoj motion

Drfl.set_safety_mode(SAFETY_MODE_AUTONOMOUS, SAFETY_MODE_EVENT_STOP);

If you Drfl.set_safety_mode(SAFETY_MODE_AUTONOMOUS, SAFETY_MODE_EVENT_MOVE); After using the command, the robot LED will start blinking. At this time, if the servoj command is used, the robot operates normally.

tf-maam, 2023-03-02:
Thank you very much for the detailed explanation. The problem still occurs using *ROS services and topics* but less frequently. 

The *dsr_control/dsr_hw_interface* respectively *dsr_control/dsr_control_node* does provide any ROS service to set the safety mode. Is it possible to implement such a service, or is it rather recommended to use the API calls directly in a ROS program?

doosan-robotics, 2023-03-03:
@tf-maam 
The **set_safety_mode** command has not been added to rosservice yet. Currently, you have to call the function through an API call.

Jeeva002, 2023-03-03:
During a motion using the movejx and movej command  the motion aborts with the undocumented alarm level : 2 group : 5 index : 7056 causing a SAFE_OFF 2,then leading to the alarms 7057 and 7069 respectively

Could you please give details on error code 7056 ,7057 , 7069 and a possible solution to overcome this problem.

doosan-robotics, 2023-03-08:
@Jeeva002 
7056: Situations where safety events do not match

7057: Joint angle exceeds safe range

7069: In case the encoder value exceeds the safe range