DoosanRobotics/doosan-robot #129: Joint speed commands do not work over speedj_rt_stream in ROS

tf-maam, 2023-01-30:
Using the `speedj_rt_stream` topic leads to an error in the `dsr_control_node` after the RT API was successfully initialized using the ROS services and commands over `servoj_rt_stream` worked fine with the same trajectory. The associated alarm message suggests that the desired position is not defined, although neither the message [`dsr_msgs/msg/ServoJRTStream.msg`](https://github.com/doosan-robotics/doosan-robot/blob/noetic-devel/dsr_msgs/msg/ServoJRTStream.msg) nor via the API function `speedj_rt`  a position can be specified.

```
...
[callback OnMonitoringStateCB] current state: (1) STANDBY
[ INFO] [1675071364.522886221]: On Monitor State
[callback OnMonitoringStateCB] current state: (2) MOVING
[ INFO] [1675071364.524000973]: On Monitor State
[callback OnMonitoringStateCB] current state: (1) STANDBY
[ INFO] [1675071364.525385078]: On Monitor State
[callback OnMonitoringStateCB] current state: (3) SAFE_OFF

[ERROR] [1675071365.042867504]: [callback OnLogAlarm]
[ERROR] [1675071365.042930147]:  level : 3
[ERROR] [1675071365.042956474]:  group : 2
[ERROR] [1675071365.042985141]:  index : 1908
[ERROR] [1675071365.043015719]:  param : 6
[ERROR] [1675071365.043044821]:  param : desired position=   -nan[deg], limit=360.000[deg]
[ERROR] [1675071365.043068131]:  param : 

[ INFO] [1675071366.299048385]: On Monitor State
[callback OnMonitoringStateCB] current state: (1) STANDBY
[ INFO] [1675071366.300156580]: On Monitor State
[callback OnMonitoringStateCB] current state: (3) SAFE_OFF
```

The same motion, provided in the RT API example, works fine on this robot.
 Could you please give some details this error and a possible solution to overcome this problem.

doosan-robotics, 2023-03-02:
@tf-maam 
It appears that the position for the 6th axis is not defined. Did you use the same source code as the API example?
We will reply again after checking.

tf-maam, 2023-03-02:
Here, I'm using the **ROS services and topics** for the RT interface. The message type [`SpeedJRTStream`](https://github.com/doosan-robotics/doosan-robot/blob/noetic-devel/dsr_msgs/msg/SpeedJRTStream.msg) to the topic `speedj_rt_stream` does not provide any field to specify an axis position. Only axis velocity, acceleration and time can be specified. 
Also in the hardware interface implementation for **ROS** only the `speedj_rt` command is called with these parameters, identically to the corresponding line in the API example (`Drfl.speedj_rt(tra.vel, tra.acc, st);`).
https://github.com/doosan-robotics/doosan-robot/blob/aa56151e558696958f5cd9debb18427b9e7782b7/dsr_control/src/dsr_hw_interface.cpp#L1282
So how is it possible to specify any axis position to avoid this error?

In our code (see the attached [Python script](https://github.com/doosan-robotics/doosan-robot/files/10873675/Doosan_ROS_speedj_rt_stream_issue.md), but also the C++ implementation shows the same behavior)  the robot is moved to home position before starting the RT control. Also sending a command message to `servo_rt_stream` with the valid initial axis positions before starting the speed controlled motion by sending [`SpeedJRTStream`](https://github.com/doosan-robotics/doosan-robot/blob/noetic-devel/dsr_msgs/msg/SpeedJRTStream.msg) messages to the topic `speedj_rt_stream` had no effect. Using the same program to command the motion via  [`ServoJRTStream`](https://github.com/doosan-robotics/doosan-robot/blob/noetic-devel/dsr_msgs/msg/ServoJRTStream.msg) messages to the topic `servoj_rt_stream` worked quite well, except the motion aborts throwing *error code 7056* as described in #128.
We also tested the identical C++ implementation, but using the direct API calls instead of the **ROS service and topic** based implementation, and there also the `speedj_rt` command was working.

doosan-robotics, 2023-03-03:
@tf-maam 
```
void DRHWInterface::speedjRTCallback(const dsr_msgs::SpeedJRTStream::ConstPtr& msg){

        std::array<float, NUM_JOINT> target_vel;
        std::copy(msg->vel.cbegin(), msg->vel.cend(), target_vel.begin());
        std::array<float, NUM_JOINT> target_acc;
        std::copy(msg->acc.cbegin(), msg->acc.cend(), target_acc.begin());
        int time = msg->time; /// invalid type (int -> float)

        Drfl.speedj_rt(target_vel.data(), target_acc.data(), time);
    }
```

I found a bug in the implementation of speed_rt_stream . time should not be set to an int. We will fixed soon.
Thank you for your reporting.

doosan-robotics, 2023-03-14:
@tf-maam 
Sorry to late. We modify the real time control method today. The time parameter type was fixed int to float.
Please check again. Thank you.