DoosanRobotics/doosan-robot #131: Unable to move robot using compliance task with MoveIt

Guillaume-Syc, 2023-02-22:
Hi there,
I'm having trouble moving the Doosan M1013 robot using the compliance function from the service "/dsr01m1013/force/task_compliance_ctrl" in contact task. When I call this service before or after the MoveIt movement planner, it plans the trajectory but doesn't actually move. Instead, I get the following error message:

state[TASK_COMPLIANCE_CONTROL] rejected event[eMoveSJ] doosan

Is there an example or tutorial available that shows how to properly use the Doosan compliance function with MoveIt? I would appreciate any guidance or advice on how to get this working. 
Thanks in advance!

doosan-robotics, 2023-02-23:
@Guillaume-Syc 
It means MOVESJ motion cannot use with task_compliance_ctrl.
For force and compliance control, user must use the Linear Motion such as MOVEL, MOVESX.
Actually, our moveit interface using MOVESJ motion for follow trajectory.

foreverbala, 2023-03-04:
@doosan-robotics 
Can I use AMOVEJX motion for task compliance and force control.
Because in this motion we are giving input as posx values. It moves as same as aMOVEL.
Is amovejx motion is linear or not?

doosan-robotics, 2023-03-06:
@foreverbala 
movejx looks like a linear motion, but it is actually a joint motion. Since it moves to the target point through joint motion, force control cannot be performed.

foreverbala, 2023-03-06:
@doosan-robotics 
Thanks for the response.

Let me explain my query.
The reason I am using amovejx motion command is to avoid singularity. Because even though it looks like linear motion it doesn't guaranties linear movement all the time. It changes to non-linear movement at the regions of singularity. 
And at the same time, I also want to use force-compliance commands. For that, the remaining option is amovel motion command by which I can do force-compliance motions. Is there is any option to use force-compliance commands and singularity avoidance commands in linear motions? Or any options to avoid singularities in linear motion. 
You have given three commands for singularity handling such as singularity avoidance, path first mode and variable velocity mode. But these are not working on real time controls. Can you guide how to use these commands? My aim is to use both force-compliance and singularity handling with a linear motion.

doosan-robotics, 2023-03-10:
@foreverbala 
Users can avoid singularity through the set_singularity_handling command.
Are you currently using real-time control commands?

foreverbala, 2023-03-11:
@doosan-robotics 
yes . I am currently using real-time control commands. I am using linear motion commands like movel, amovel .During this in order to avoid singularity I used  set_singularity_handling commands like singularity avoidance, path first mode and variable velocity. But it doesn't work.Below the code is given.

`while rospy is not shutdown:      
         
                movej(p1, vel=60, acc=30)
                movej(p1, vel=50, acc=50)
                movel(x2, velx, accx)
                **set_singularity_handling (DR_AVOID)**
                task_compliance_ctrl(stx = [500, 500, 500, 100, 100, 100],time = 0.1)
                # # set_stiffnessx([100, 100, 300, 100, 100, 100])
                fd = [0, 0, 10, 0, 0, 10]
                fctrl_dir= [1, 1, 1, 1, 1, 1]
                set_desired_force(fd, dir=fctrl_dir, time=1.0, mod=DR_FC_MOD_REL)
                movej(p3, vel=60, acc=30)
                movej(p4, vel=60, acc=30)
                **set_singularity_handling (DR_AVOID)**            
                movel(x2, velx, accx)`

Guillaume-Syc, 2023-04-12:
Hello,
I'm trying to use the doosan compliance. Currently, I use a moveL (relative 10cm) to go down (the robot enters in contact with the surface) and then I use a moveL to go up (relative: 15cm).  I use rosservice to execute the move. 
As the robot can't access the first target (it collides with the surface), I can't send the second moveL upwards and the robot triggers an alarm. 
```
movel(x1, time=2, mod=DR_MV_MOD_REL)  
task_compliance_ctrl()
movel(x2, time=2, mod=DR_MV_MOD_REL)
release_compliance_ctrl()
``` 
How can I use it?