DoosanRobotics/doosan-robot #134: When planning the robotic arm, the first and second joints are often deviated

xwn-github, 2023-03-02:
您好，我在使用ros melodic版本的doosan-robot功能包时，每次使用moveit（pkg: dsr_launcher dsr_moveit）运动机器人时，第一关节和第二关节经常偏离，终端报错：start point deviates from current robot state more than 0.01.
分析原因：每次用moveit运动都会出现，可能是发送目标点之后机械臂运动到位了，但是在ros中机械臂还在原位。就是每次发送目标点，机械臂实际运动之后都会出现这个错误。

Hello, when I use the ros melodic version of the doosan-robot function pack, every time I use the moveit (pkg: dsr_launcher dsr_moveit) motion robot, the first and second joints often deviate, and the terminal error: start point deviates from current robot state more than 0.01.
Analysis reason: Every time the movement with MOVEIT will appear, it may be that the robotic arm moves in place after sending the target point, but the robotic arm is still in place in ROS. That is, every time the target point is sent, this error will occur after the actual movement of the robotic arm.

[image]

doosan-robotics, 2023-03-06:
@xwn-github 
The joint state of the Doosan robot is updated every 100 ms, and in the case of moveit, it is operated using the movesj (spline joint) command.
I suspect an imbalance between planning time and joint state update.

xwn-github, 2023-03-07:
Is there any solution?

doosan-robotics, 2023-03-07:
@xwn-github 
There is no clear solution. It seems to be a problem caused by the difference between the actual location after planning and the location recognized on rviz.

xwn-github, 2023-03-07:
Yes, connect a real robot, model is M1013

doosan-robotics, 2023-03-07:
@xwn-github 
Let's run a test to analyze the exact reason.
Did you perform motion through an interactive marker?

xwn-github, 2023-03-07:
是的

doosan-robotics, 2023-03-14:
@xwn-github 
As a result of the test, moveit worked as an interactive marker without any problems on our side.

doosan-robotics, 2023-03-28:
@xwn-github 
If you encounter additional issues, please reopen the issue or create a new one.

closed 2023-03-28 by doosan-robotics

charleswilmot, 2024-01-19:
Chances are that the problem comes from here:

```cpp
    void JointTrajectoryAction::trajectoryCallback(const control_msgs::FollowJointTrajectoryGoalConstPtr &goal)
    {
        ROS_INFO("callback: Trajectory received");
        ROS_INFO("  goal->trajectory.points.size() =%d", (int)goal->trajectory.points.size());           //=10 가변젹
        ROS_INFO("  goal->trajectory.joint_names.size() =%d", (int)goal->trajectory.joint_names.size()); //=6

        float preTargetTime = 0.0;
        float targetTime = 0.0;

        int nCntTargetPos = goal->trajectory.points.size();

        float fTargetPos[nCntTargetPos][NUM_JOINT] = {
            0.0,
        };
        // if (nCntTargetPos > MAX_SPLINE_POINT)
        // {
        //     ROS_INFO("DRHWInterface::trajectoryCallback over max Trajectory (%d > %d)", nCntTargetPos, MAX_SPLINE_POINT);
        //     as_.setAborted(result_);
        //     return;
        // }

        ros::Time begin = ros::Time::now();

        for (int i = 0; i < nCntTargetPos; i++) //=10
        {
            std::array<float, NUM_JOINT> degrees;
            ros::Duration d(goal->trajectory.points[i].time_from_start);

            // ROS_INFO("  goal->trajectory.points[%d].time_from_start = %7.3%f",i,(float)goal->trajectory.points[i].time_from_start );

            targetTime = d.toSec();
            /// ROS_INFO("[trajectory] preTargetTime: %7.3f", preTargetTime);
            /// targetTime = targetTime - preTargetTime;
            /// preTargetTime = targetTime;
            /// ROS_INFO("[trajectory] time_from_start: %7.3f", targetTime);

            for (int j = 0; j < goal->trajectory.joint_names.size(); j++) //=6
            {
                // ROS_INFO("[trajectory] %d-pos: %7.3f", j, goal->trajectory.points[i].positions[j]);
                /* todo
                get a position & time_from_start
                convert radian to degree the position
                run MoveJ(position, time_From_start)
                */
                degrees[j] = rad2deg(goal->trajectory.points[i].positions[j]);

                fTargetPos[i][j] = degrees[j];
            }

            ros::Duration step_duration = d - (ros::Time::now() - begin);
            float blending_radius = 50;

            if (as_.isPreemptRequested() || !ros::ok())
            {
                ROS_INFO("%s: Preempted", action_name_.c_str());
                // set the action state to preempted
                as_.setPreempted();
                return;
            }
            ROS_INFO("[trajectory] [%02d : %.3f : %.3f] %7.3f %7.3f %7.3f %7.3f %7.3f %7.3f", i, targetTime, step_duration.toSec(), rad2deg(goal->trajectory.points[i].positions[0]), rad2deg(goal->trajectory.points[i].positions[1]), rad2deg(goal->trajectory.points[i].positions[2]), rad2deg(goal->trajectory.points[i].positions[3]), rad2deg(goal->trajectory.points[i].positions[4]), rad2deg(goal->trajectory.points[i].positions[5]));

            Drfl.MoveJAsync(degrees.data(), 50, 50, step_duration.toSec()+0.25, MOVE_MODE_ABSOLUTE, BLENDING_SPEED_TYPE_OVERRIDE);

            // ros::Time::sleepUntil(begin + d - ros::Duration(0.5));
            ros::Time::sleepUntil(begin + d);
        }
        Drfl.MoveWait();
        // ROS_INFO("CALLING MOVESJ");
        // Drfl.movesj(fTargetPos, nCntTargetPos, 0.0, 0.0, targetTime, (MOVE_MODE)MOVE_MODE_ABSOLUTE);
        // ROS_INFO("CALLED MOVESJ");
        /*
        for(int i = 0; i < NUM_JOINT; i++){
            ROS_INFO("[]::cmd %d-pos: %7.3f", i, joints[i].cmd);
            cmd_[i] = joints[i].cmd;
        }
        */
        as_.setSucceeded(result_);
    }
```