DoosanRobotics/doosan-robot #135: servo turned off while using movejx and movej motion command

Jeeva002, 2023-03-03:
During a motion using the movejx and movej command the motion aborts with the undocumented alarm level : 2 group : 5 index : 7056 causing a SAFE_OFF 2,then leading to the alarms 7057 and 7069 respectively

Could you please give details on error code 7056 ,7057 , 7069 and a possible solution to overcome this problem.

doosan-robotics, 2023-03-06:
@Jeeva002 
In general, when using movejx or movej, it is configured so that the 5.7056 error does not occur.
A hardware problem is suspected. Can you show the example code you used?

Jeeva002, 2023-03-06:
`#!/usr/bin/env python3
# -*- coding: utf-8 -*-
import rospy
import sys
import os
import statistics
from sensor_msgs.msg import Joy

sys.dont_write_bytecode = True
sys.path.append( os.path.abspath(os.path.join(os.path.dirname(__file__),"../../../../common/imp")) ) 

ROBOT_ID     = "dsr01"        
ROBOT_MODEL  = "a0509"  

import DR_init
DR_init.__dsr__id = ROBOT_ID
DR_init.__dsr__model = ROBOT_MODEL
from DSR_ROBOT import *

nameSpaceObj = CDsrRobot(ROBOT_ID, ROBOT_MODEL)

robomovement=[0,0,0,0,0,0]

home_pos=posj(0,0,0,0,0,0)
ready_pos=posj(0,-10,100,0,90,0)

movement_flag = True
class roboCore:

    def __init__(self):
    
      self.countx = 0 
      self.abstorel_count=0
     
      self.robovelocity=40
      self.roboacceleration=100
  

 
      self.pos_first_flag=False
      self.pos_second_flag=False
  
      
    def axes_function(self, joy_x,joy_y,joy_z,joy_rx,joy_ry,joy_rz):
           
      
           self.rob_x = [0]
           self.rob_y = [0]
           self.rob_z = [0]
           self.rob_rx = [0]
           self.rob_ry = [0]
           self.rob_rz = [0]

           self.rob_x.append(joy_x)
           self.rob_y.append(joy_y)
           self.rob_z.append(joy_z)
           self.rob_rx.append(joy_rx)
           self.rob_ry.append(joy_ry)
           self.rob_rz.append(joy_rz)

           self.countx += 1 
   

           if self.countx == 8:
            
                 x_average=statistics.mean(self.rob_x)
                 y_average=statistics.mean(self.rob_y)
                 z_average=statistics.mean(self.rob_z)
                 rx_average=statistics.mean(self.rob_rx)
                 ry_average=statistics.mean(self.rob_ry)
                 rz_average=statistics.mean(self.rob_rz)
                 if self.pos_first_flag == True:
                 
                      ABS_TO_REL =get_current_tool_flange_posx()
                  
                      robomovement[0]=ABS_TO_REL[0]+round(x_average,1)
                      robomovement[1]=ABS_TO_REL[1] +round(y_average,1)
                      robomovement[2]=ABS_TO_REL[2] +round(z_average,1)
                      robomovement[3]=ABS_TO_REL[3] +round(rx_average,1)
                      robomovement[4]=ABS_TO_REL[4] +round(ry_average,1)
                      robomovement[5]=ABS_TO_REL[5] +round(rz_average,1)

                      nameSpaceObj.amovejx(robomovement,self.robovelocity,self.roboacceleration,sol=2)
                     
                      self.pos_second_flag = True
                      self.pos_first_flag=False
                
                      
                      self.countx=0
               
                      
                      self.rob_x.clear()
                      self.rob_y.clear()
                      self.rob_z.clear()
                      self.rob_rx.clear()
                      self.rob_ry.clear()
                      self.rob_rz.clear()
                     
              
                  
                 elif self.pos_second_flag == True: 
            

                      robomovement[0]=round(robomovement[0],1) + round(x_average,1)
                      robomovement[1]=round(robomovement[1],1) + round(y_average,1)
                      robomovement[2]=round(robomovement[2],1) + round(z_average,1)
                      robomovement[3]=round(robomovement[3],1) + round(rx_average,1)
                      robomovement[4]=round(robomovement[4],1) + round(ry_average,1)
                      robomovement[5]=round(robomovement[5],1) + round(rz_average,1)
                     
             
                      nameSpaceObj.amovejx(robomovement,self.robovelocity,self.roboacceleration,sol=2)
                      
                 
                    
                     
                      self.countx=0

                    
                  
                      self.rob_x.clear()
                      self.rob_y.clear()
                      self.rob_z.clear()
                      self.rob_rx.clear()
                      self.rob_ry.clear()
                      self.rob_rz.clear()
                    

robo_obj = roboCore()

def callback(data):
     

          joy_x=round(80*data.axes[0],1)
          joy_y=round(80*data.axes[1],1)
          joy_z=round(80*data.axes[2],1)
          joy_rx=round(80*data.axes[3],1)
          joy_ry=round(80*data.axes[4],1)
          joy_rz=round(80*data.axes[5],1)
         
     
          robo_obj.axes_function(joy_x,joy_y,joy_z,joy_rx,joy_ry,joy_rz)

        
        
def start():
        rospy.Subscriber("joy", Joy,callback)
        rospy.init_node('joyfilter')   
         
   
`

Jeeva002, 2023-03-06:
@doosan-robotics The above code is for to control the doosan a0509 robot with logitech extreme3d pro joystick, I call the function def start() from a main file,  program will send the command to move the robot after it has received the joy data from a joy package ,approximately i tried to move the robot 2 to 4 times in second each movement consist less than 60 mm ,  .

doosan-robotics, 2023-03-07:
@Jeeva002 
The movej and movejx motions included in ROS are not suitable for joystick integration. It is presumed that an error such as 5.7056 occurred because a command was input while the safety status was changing. It is recommended to use rostopic commands such as servoj_stream or jog_multi.

Jeeva002, 2023-03-07:
@doosan-robotics  jog_multi topic is not functioning properly ,i posted that issues here #136 , is jog_multi function is officially released?

doosan-robotics, 2023-03-08:
@Jeeva002 
There are no plans for an official release yet. For more than jog_multi and singularity commands, I will answer again after testing.

Jeeva002, 2023-03-08:
@doosan-robotics okay,update here the result after Testing . 
Thank you for your response🤝

doosan-robotics, 2023-03-28:
@Jeeva002 
If you encounter additional issues, please reopen the issue or create a new one.

closed 2023-03-28 by doosan-robotics