DoosanRobotics/doosan-robot #136: set_singularity_handling(mode) and JogMulti.srv are not working in doosan a0509

Jeeva002, 2023-03-06:
i tried to move the robot using jogmulti ,example : i tried to move the robot in x direction the robot moves maximum in x direction ,what ever the the value i tried to move in any direction it moves maximum in that direction, i tried this too(-1,+1)

set_singularity_handling command is also not functioning  
Could you please give details on the above issue and a possible solution to overcome this problem.

doosan-robotics, 2023-03-07:
@Jeeva002 
It is recommended to use topic rather than jog_multi service.
Do you have such a problem while using topic?

Jeeva002, 2023-03-07:
yes,iam having same issues in both jogmulti topic and jog multi service

doosan-robotics, 2023-03-14:
@Jeeva002 
As a result of the test, it was confirmed that the jog_multi command is working normally. Is it a problem that it keeps moving in the input direction when using jog_multi? jog_multi continues to move as much as speed in the input direction[x, y, z, r, p, y] as velocity control.

Jeeva002, 2023-03-14:
@doosan-robotics  Is it working properly in A0509 Robot ? YES, it keeps moving continously in input direction .If i use  +1 for x positive direction it moves maximum in x postive like this jog_multi function works?

doosan-robotics, 2023-03-14:
@Jeeva002 
The jog_multi command supports online trajectory. If another input comes in while inputting, it moves in the changed direction. If you want to stop the jog_multi command, enter 0 for speed or direction.
The set_singularity_handling command was also confirmed to work normally. However, I will check whether the command affects the jog_multi command and answer again.

doosan-robotics, 2023-04-04:
If another issue arises, please reopen the issue or create another issue.

closed 2023-04-04 by doosan-robotics