DoosanRobotics/doosan-robot #138: Controller Spawner couldn't find the expected controller_manager

xwn-github, 2023-03-15:
我在运行”roslaunch dsr_launcher single_robot_rviz.launch“时，RVIZ界面显示模型错误：
[image]
终端警告：Controller Spawner couldn't find the expected controller_manager 
[image]
我不知道是什么原因，请您帮忙分析原因。

doosan-robotics, 2023-03-16:
@xwn-github 
Are you trying to test in a virtual environment?
If so, you need to run our controller simulator with sudo privileges (the emulator is subject to revision). Please note that the emulator is an old version.

After running the controller emulator as shown below, run the launch file again.

cd doosan-robot/common/bin/DRCF
sudo ./DRCF

If the emulator ran without errors, run the launch file.

xwn-github, 2023-03-16:
是的，我是在虚拟环境中进行测试。
我尝试过了你的这个方法，结果是一样的。

[image]

doosan-robotics, 2023-03-16:
[image]
在我这边，它是这样工作的。是否正常运行？

xwn-github, 2023-03-16:
能把你这个源码发我吗？我对比一下代码有什么不同。我的邮箱是983078325@qq.com
非常感谢！

xwn-github, 2023-03-16:
我重装Ubuntu系统，重装ROS，重新构建ROS功能包，机器人模型还是显示不出来。
能把你这个源码发我吗？我对比一下代码有什么不同。我的邮箱是983078325@qq.com
非常感谢！

xwn-github, 2023-03-17:
我反馈两个问题，希望您更新维护一下源代码。
问题1：
运行dsr_control 包下 dsr_moveit.launch 启动文件时
终端报错：
RLException: Invalid roslaunch XML syntax: mismatched tag: line 25
The traceback for the exception was written to the log file
原因：检查 dsr_moveit.launch文件的标签，<param>标签没有结束标志/。修改后运行dsr_moveit.launch成功。
[image]

问题2：
运行gazebo相关启动文件时，出现No p gain specified for pid问题
终端报错：
[ERROR] [1678204846.487352488, 0.128000000]: No p gain specified for pid.  Namespace: /dsr01/gazebo_ros_control/pid_gains/joint1
原因：PID配置文件调用语句被注释掉；joint1-joint6的PID参数被注释掉；PID配置文件YAML格式错误（换行、空格）。
解决办法：
（1）找到dsr_bash.launch文件（路径：doosan_robot/dsr_gazebo/launch/），看到调用PID配置从YAML文件到参数服务器的语句被注释掉了，取消注释。

<group ns="$(arg ns)">
    <rosparam file="$(find dsr_gazebo)/config/gazebo_ros_control_params.yaml" command="load"/>
</group>
（2）找到gazebo_ros_control_params.yaml文件（路径：doosan_robot/dsr_gazebo//config/），修改如下：
gazebo_ros_control:
   pid_gains:
      joint1: {p: 500.0, i: 0, d: 85}
      joint2: {p: 1000.0, i: 0, d: 100}
      joint3: {p: 1000.0, i: 0, d: 100}
      joint4: {p: 500.0, i: 0, d: 85}
      joint5: {p: 600.0, i: 0, d: 85}
      joint6: {p: 500.0, i: 0, d: 85}
        
      front_left_wheel: {p: 500.0, i: 0.0, d: 1.0}
      front_right_wheel: {p: 500.0, i: 0.0, d: 1.0}
      rear_left_wheel: {p: 500.0, i: 0.0, d: 1.0}
      rear_right_wheel: {p: 500.0, i: 0.0, d: 1.0}
加载PID配置后，运行gazebo不会再出现以上报错，但是会引发另一个问题：gazebo里的机器人模型一直在抖动。原因是PID参数不合理，需要调节参数。猜想官方不用PID配置的原因就在此。尝试调试PID参数，效果有好有坏，但不能完全解决抖动问题，PID没有调试到最佳值。

duygiang1162, 2024-03-27:
How to fix this problem? Please give me some advice?

vyasakash231, 2024-04-03:
> [image] 在我这边，它是这样工作的。是否正常运行？

I'm still getting the same error after this,
[image]