DoosanRobotics/doosan-robot #139: Servo on/off from ros control

Bhanu5302, 2023-03-16:
How Can we turn on/off servo through ros control

doosan-robotics, 2023-03-16:
@Bhanu5302 
You can use https://github.com/doosan-robotics/doosan-robot/blob/noetic-devel/dsr_msgs/srv/system/SetRobotControl.srv for servo on. But servo off command is not implemented yet. We are going to develop.

doosan-robotics, 2023-03-16:
@Bhanu5302 
If parameter 3 is entered in the set_robot_control service, the servo-on function is activated.

Bhanu5302, 2023-03-16:
We need servo off function also for our application. So, I request you to develop that function as soon as possible

Bhanu5302, 2023-03-16:
Thank you @doosan-robotics  for the information. Hope you will update the servo off function soon

closed 2023-03-16 by Bhanu5302

MLDario, 2024-08-30:
The only way I found to do that is by connecting and disconnecting the robot using subprocesses in Python.