DoosanRobotics/doosan-robot #140: A h2017_moveit_controllse_manager.launch.xml file is wrong name

nghiank97, 2023-03-19:
When I run a command **roslaunch dsr_launcher dsr_moveit.launch model:=h2017 mode:=virtual**. A error appear

[image]

I found the **h2017_moveit_controllse_manager.launch.xml** file is wrong name. 

[image]

It have to **h2017_moveit_controller_manager.launch.xml**. i edited and i success

doosan-robotics, 2023-03-19:
Thank you for finding the wrong file name. Could you pull request the modified file?

doosan-robotics, 2023-03-23:
Hi, I fixed the file name, please check and pull the branch, thanks!

closed 2023-03-23 by doosan-robotics